riscv/riscv-crypto #16: Instruction encodings.

ben-marshall, 2020-06-30:
Placeholder issue for discussing instruction encodings.

Currently we are waiting for guidance from the foundation for how this process will work.

grnewell, 2020-07-01:
Ken Dockser has volunteered to help the TG with the initial proposal for encodings.

VincenttttWu, 2021-02-24:
hello everyone, Notice that almost all encodings of Instruction in K Ext are located SYSTEM, OP & OP-IMM field in ‘RISC-V base opcode map’. Could you please tell me does it will be changed in the future.  Moes it possible to change K Ext into reserved field in ‘RISC-V base opcode map’?

ben-marshall, 2021-02-24:
Hi @VincenttttWu 

The K extension is currently going through the opcode consistency and review process, meaning there is a chance the opcodes/encodings will change based on the recommendations which come out of that. We are *hoping* this won't be the case, but it's possible. You should treat the opcodes as if they will change for now.

I'm afraid I don't understand your question about changing the opcode into a reserved field of the base opcode map?

Cheers,
Ben

VincenttttWu, 2021-02-24:
Cuz I am working for implement MC layer of LLVM for K Ext. So I need to decide should I use a new OPcode variable for K Ext or original one(SYSTEM, OP & OP-IMM etc.). although they have same value in this stage.

Sincerely yours,
Vincent

ben-marshall, 2021-02-24:
Hi @VincenttttWu 
Ah I see. I'm no LLVM expert so you are the best judge. I expect that whatever the K ext encodings end up being, they will be in an original opcode space like SYSTEM/OP/OP-IMM, since that's the current policy for allocating new instructions. They may change *within* those spaces, but I highly doubt they will be in a new greenfield / reserved space. I hope that helps?
Cheers,
Ben

VincenttttWu, 2021-02-24:
Hello @ben-marshall 

Sure, it's helps a lot. Thank you.

Sincerely yours,
Vincent

closed 2021-06-04 by ben-marshall