riscv/riscv-crypto #17: Vector extension: Carryless multiply.

ben-marshall, 2020-06-30:
A discussion / proposal for a carry-less multiply instruction to be added to the vector cryptography extensions.

- This is motivated by the [GHASH part of GCM](https://en.wikipedia.org/wiki/Galois/Counter_Mode).

- The instructions can work similarly to the [existing vector integer multiply instructions](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1210-vector-single-width-integer-multiply-instructions), something like:

  ```
  vclmul.vv  vd, vs2, vs1, vm // Vector-vector lo
  vclmul.vs  vd, vs2, vs1, vm // Vector-scalar lo
  vclmulh.vv vd, vs2, vs1, vm // Vector-vector hi
  vclmulh.vs vd, vs2, vs1, vm // Vector-scalar hi
  ```

  Where (for the Crypto extension) `SEW=128`. Other extensions might want to implement other SEW values, but for Crypto, we only need `SEW=128`?

Two further issues:

- In [Markku's analysis](https://github.com/scarv/riscv-crypto/blob/master/doc/supp/gcm-mode-cmul.adoc) for the scalar implementation of GCM, he [used](https://github.com/mjosaarinen/lwaes_isa/blob/master/gcm_rv64b_gfmul.c#L17) the Bitmanip "`grev`" instruction to reverse the bits in every byte of the input. I can't see an existing instruction in the V-spec which does this efficiently. Adding an instruction like

  ```
  vbrev.vv vrt    // Bit reverse, destructive.
  ```

  Which reverses the bits in every scalar element based on `SEW`, and requiring support for `SEW=8` for the crypto extension seems reasonable?

  There are other (non-obvious) [ways to do bit reversal quickly](https://graphics.stanford.edu/~seander/bithacks.html#ReverseByteWith64BitsDiv) which look amenable to vectorising, which we can fall back on.

- The reduction operation. Again, based on Markku's analysis for the scalar stuff, this [can be done](https://github.com/mjosaarinen/lwaes_isa/blob/master/gcm_rv64b_gfmul.c#L81) efficiently (in terms of instruction counts at-least) using `vclmul[h].vs` instructions and `vxor`, so a dedicated instruction might not be needed?

  I'm not sure how the calculus about reduction by multiplication or reduction by shift/xor applies in the vector world so much? Intuition says that the same calculus for the scalar world will apply to the vectors, in which case, it's up to implementers to optimise for themselves. The [background](https://github.com/scarv/riscv-crypto/blob/master/doc/supp/gcm-mode-cmul.adoc#reduction-via-shifts-or-with-multiplication) being that the fastest method depends on the speed of the carry-less multiply operation relative to shifts and xors.

grnewell, 2020-07-01:
Can we separate these topics into different GitHub issues?

ben-marshall, 2020-07-02:
Sure, I've split out the bit-reversal stuff into #18. I left the reduction discussion in because I'm not sure how to dis-entangle it. I hope that's alright.

ben-marshall, 2020-07-03:
**WIP** - This comment is still being edited!
---

Following the TG meeting on June 2'nd 2020, these options are put forward for comparison. There are two broad categories:
- Hi/Lo instructions.
- Widening instructions.

There are also two orthogonal questions which apply to both categories:
- Should we include a multiply-accumulate?
- Which values of `SEW` should the crypto extension *require* support for?

### Option 1: Hi/Lo instructions

```
vclmul.vv   vrd, vrs1, vrs2, vm  // vrd[i] = vrs1[i] * vrs2[i] (SEW*SEW -> low  SEW)
vclmulh.vv  vrd, vrs1, vrs2, vm  // vrd[i] = vrs1[i] * vrs2[i] (SEW*SEW -> high SEW)
vclmul.vs   vrd, vrs1,  rs2, vm  // vrd[i] = vrs1[i] *  rs2    (SEW*SEW -> low  SEW)
vclmulh.vs  vrd, vrs1,  rs2, vm  // vrd[i] = vrs1[i] *  rs2    (SEW*SEW -> high SEW)
```
- These instructions work analogously to the base vector spec [Single-Width Integer Multiply Instructions](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1210-vector-single-width-integer-multiply-instructions)
- The `vclmul.*` instructions compute the low `SEW` bits of the `SEW*SEW`-bit multiply
- The `vclmulh.*` instructions compute the high `SEW` bits.

### Option 2: Widening Instructions, with lo

```
vwclmul.vv   vrd, vrs1, vrs2, vm  // vrd[i] = vrs1[i] * vrs2[i] (SEW*SEW -> 2*SEW)
vwclmul.vs   vrd, vrs1,  rs2, vm  // vrd[i] = vrs1[i] *  rs2    (SEW*SEW -> 2*SEW)
vclmul.vv    vrd, vrs1, vrs2, vm  // vrd[i] = vrs1[i] * vrs2[i] (SEW*SEW -> low SEW)
vclmul.vs    vrd, vrs1,  rs2, vm  // vrd[i] = vrs1[i] *  rs2    (SEW*SEW -> low SEW)
```

- These instructions work analogously to the base vector spec [Widening Integer Multiply Instructions](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1212-vector-widening-integer-multiply-instructions)
- The `vwclmul.*` instructions compute the `2*SEW`-bit result of the `SEW*SEW`-bit multiplication.
  - `EEW=  SEW` for `vrs1`, `vrs2` and `rs2`
  - `EEW=2*SEW` for `vrd`
  - The widening instructions are used for the multiplication part of the GHASH operation.
- The `vclmul.*` instructions work identically to the ones in Option 1 and are used for the reduction.
- Questions:
  - Does using the widening instructions for the multiplication and non-widening for the reduction require a change of `SEW` value at any point?
  - When mixing widening and non-widening, do the `2*SEW` result elements of the widening instructions end up in the right places to easily
---

Cross cutting questions:

### Which values of `SEW` to require?

- The critical case for the vector crypto extension is `SEW=128`.
- Requiring `SEW=128` may be burdensome and face resistance.
- An alternative would be to require support for `SEW=XLEN`
  - This would make the vector code extremely similar to the [scalar code](https://github.com/mjosaarinen/lwaes_isa/blob/master/gcm_rv64b_gfmul.c#L25)
  - Implementers would be free to support `SEW >= XLEN`.

### Should we include carry-less multiply-add?

- For supported values of `SEW < 128`, the `vclmac*` instructions become particularly useful as they fuse summing `vxor` operations.
- Including multiply accumulate would add (a subset of) the following instructions:
```
vclmacc.vv    vrd, vrs1, vrs2, vm  // vrd[i] += vrs1[i] * vrs2[i]
vclmacch.vv   vrd, vrs1, vrs2, vm  // vrd[i] += vrs1[i] * vrs2[i]
vclmacc.vs    vrd, vrs1,  rs2, vm  // vrd[i] += vrs1[i] *  rs2
vclmacch.vs   vrd, vrs1,  rs2, vm  // vrd[i] += vrs1[i] *  rs2
vwclmacc.vv   vrd, vrs1, vrs2, vm  // vrd[i] += vrs1[i] * vrs2[i]
vwclmacch.vv  vrd, vrs1, vrs2, vm  // vrd[i] += vrs1[i] * vrs2[i]
vwclmacc.vs   vrd, vrs1,  rs2, vm  // vrd[i] += vrs1[i] *  rs2
vwclmacch.vs  vrd, vrs1,  rs2, vm  // vrd[i] += vrs1[i] *  rs2

```
- These instructions work analogously to the base vector spec [Single-Width Integer Multiply-Add Instructions](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1213-vector-single-width-integer-multiply-add-instructions) and [Vector Single-Width Integer Multiply-Add Instructions](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1213-vector-single-width-integer-multiply-add-instructions).
- Their inclusion removes the need for `vxor` instructions.
- For the multiplication part of GHASH, only the `*.vv` versions are needed?
- For the reduction, it's would only the `*vs` versions would be used?

ben-marshall, 2020-07-06:
Added an RFC under the supplementary information in the `dev/next-release` branch.
See https://github.com/scarv/riscv-crypto/blob/dev/next-release/doc/supp/vector-clmul-rfc.md

ben-marshall, 2020-07-16:
Closed by commit 9ddd1a
Future specific issues around Vector CLMUL will have new issues raised.

closed 2020-07-16 by ben-marshall