riscv/riscv-crypto #18: Vector bit reversal

ben-marshall, 2020-07-02:
In the context of #17 (Vector extension: Carry-less Multiply) - How best to handle needing to reverse bits-in-bytes to get the input in the correct form?

- In [Markku's analysis](https://github.com/scarv/riscv-crypto/blob/master/doc/supp/gcm-mode-cmul.adoc) for the scalar implementation of GCM, he [used](https://github.com/mjosaarinen/lwaes_isa/blob/master/gcm_rv64b_gfmul.c#L17) the Bitmanip "`grev`" instruction to reverse the bits in every byte of the input. I can't see an existing instruction in the V-spec which does this efficiently. Adding an instruction like

  ```
  vbrev.vv vrt    // Bit reverse, destructive.
  ```

  Which reverses the bits in every scalar element based on `SEW`, and requiring support for `SEW=8` for the crypto extension seems reasonable?

- There are other (non-obvious) [ways to do bit reversal quickly](https://graphics.stanford.edu/~seander/bithacks.html#ReverseByteWith64BitsDiv) which look amenable to vectorising, which we can fall back on.

grnewell, 2020-07-03:
The best alternative method I have seen so far, adapted for RVV from one of the methods in the link in the main post ([this one](https://graphics.stanford.edu/~seander/bithacks.html#ReverseParallel)) , uses masking and shifting and takes 15 base vector extension instructions to reverse the order of the 8 bits in all the bytes of all the elements in a vector (not counting loading and other set-up overhead)

ben-marshall, 2020-07-28:
Andy's proposal for vector grev can be found [here](https://github.com/AndyGlew/Ri5-stuff/wiki/VGREV---Vector-Generalized-Reverse). These will be transcribed into the draft spec.

commit 30c5e8c5 (2020-08-03): spec: add specifications for vector rotate, grev.

- Closes #18 - Vector bit reversal.
- Closes #19 - Vector extension: rotation.

 On branch dev/next-release
 Your branch is ahead of 'origin/dev/next-release' by 3 commits.
   (use "git push" to publish your local commits)

 Changes to be committed:
	modified:   doc/riscv-crypto-spec.sty
	modified:   doc/tex/changelog.tex
	modified:   doc/tex/sec-scalar-bitmanip.tex
	new file:   doc/tex/sec-vector-grev.tex
	new file:   doc/tex/sec-vector-rotate.tex
	modified:   doc/tex/sec-vector.tex

 Changes not staged for commit:
	modified:   extern/riscv-g [... cut]
closed 2020-08-04 by ben-marshall