riscv/riscv-crypto #21: Crypto Implementation Profiles Questions

kgoveas-ventana, 2020-07-22:
Hi Ben, thanks for the work on the latest crypto spec. I was reading through it and had a couple of questions related to 'implementation profiles'. I wasn't sure whether to just email the group or file an issue. I decided to do the latter but let me know if you prefer the former.

   1.  Does support for an implementation profile imply an implementation need only add support for the subset of instructions corresponding to that profile  e.g sm3/sm4 instructions are not supported if only scalar profile #1 is supported?
       1. Will there be misa bits assigned to provide software with a way to discover implementation profiles supported?
  1. There seems to be an inconsistency in the Table 1 versus the ensuing description for vector profiles. The description states that Profile 1 is for "per-round" instructions but Table 1 indicates it supports Vector AES per-round and all-round. Similarly, the description for Profile 2 states that it is for "all-round" instructions but Table 1 indicates it is for Vector SHA2 Per-Round and All-Round. 

Thanks,
Kelvin

ben-marshall, 2020-07-23:
Hi Kelvin

An issue or an email is fine, depending on the question I think an email will get more widely read.

> Does support for an implementation profile imply an implementation need only add support for the subset of instructions corresponding to that profile

That's the idea yes, though the profiles currently in the draft spec are very much the start of a discussion. I think this is one of the big topics soon to be discussed at the TG meetings.

> Will there be misa bits assigned to provide software with a way to discover implementation profiles supported?

I think the plan at the moment is to use the`K` bit of `misa` to indicate that *any* of the crypto extension is supported. To find out exactly which parts of the extension are implemented, we plan to re-use what the [`tech-config`](https://lists.riscv.org/g/tech-config/) task group come up with.

> There seems to be an inconsistency in the Table 1 versus the ensuing description for vector profiles.

Ah, yes. That's a typo on my part. Thanks for pointing it out!

Cheers,
Ben

ben-marshall, 2020-07-23:
Hi @kgoveas-ventana - Can i leave it to you to close this issue? I've fixed my typo now and hopefully I answered what questions you had, even if they're mostly "thats a TBD".

If you have ant ideas / requirements around the profiles, it would be really helpful to hear them.

Cheers,
Ben

kgoveas-ventana, 2020-07-23:
Thanks for the update, Ben. I'll email you and the group any additional comments I have on the spec. 

Kelvin

closed 2020-07-23 by kgoveas-ventana