riscv/riscv-crypto #35: Overall Issue - what are the crypto feature grouops and recommended subsets

AndyGlew, 2020-08-27:
This is a parent issue that should link to several different child issues.

We want to define ... profiles? ... defined subsets of crypto features, so that e.g. code running on a Linux system can know that it is allowed to generate which instructions.  Not just which instructions, but which combinations of instrucrtions - since coding, optimization, and validation strategies may depend on certain (sub) sets of instructions all being available at the same time. 

However, beyond or  underneath such "profiles"  may lie subsets or groupings of instructions  that can be imagined as being composable,  although the compositions may be restricted  by the "profiles".  E.g. there may be composable subsets of full-round  vector encryption and decryption instructions, ve and vd.  ve and vd are the composable groups.  But the constrained sets may be only {ve} and {ve+vd}, i.e. we may see no value in supporting vd without ve.

This issue just wants to record and collect together subissues related to this overall topic.

---

I say "...profiles?..." because I am not sure what the official term for this sort of constraint subset is. At one point we were calling these profiles, but recently the term "profiles:"  has acquired a specific meaning for RISC-V  that may or may not match that older use.

---

For example -  this example would be an issue of its own, if there was any disagreement.

Public key code using GF(2^n) may only want to support two versions of the code:

1. code that users both the crypto vector VCLMUL instruction, and VGREV (specifically the VGREV versdion that reverses bits in bytes).

2.  code that uses neither.

We want to discourage implementations from providing VCLMUL without VGREV; we do not want coders to waste time writing code that is optimized for that case.

ben-marshall, 2020-08-28:
Child issues:
- #36 Crypto subsets: AES192 always optional, vs AES128 and AES256 ?
- #37 Crypto subsets: AES128 + AES256 always come together ?
- #38 Crypto subsets: {AES128+AES256}, {AES128+AES256+AES192}
- #39 Crypto subsets: should there be a profile that has both all-rounds and by-round instructions ?in same

grnewell, 2020-09-26:
I have updated the vector proposed naming strings and associated vector instruction groupings.  See [this spreadsheet](https://docs.google.com/spreadsheets/d/1LU85JcAhQ60DXPnSYtp8zOgBo5nOsSR6utJsu_mqk2g/edit?usp=sharing) for the latest.[](url)

ben-marshall, 2020-10-23:
As decided at meeting 23/oct/2020 - closing all feature-group related issues and tracking in Jira.

closed 2020-10-23 by ben-marshall