riscv/riscv-crypto #36: Crypto subsetting: AES192 always optional, vs AES128 and AES256 ?

AndyGlew, 2020-08-27:
AES comes in flavors 128, 192, and 256.

AES192 is  hardly ever used, if at all.

PROPOSAL - POR

Therefore, we propose that AES192 be always optional.  

Implementations are free not to implement AWES192 at all.

But... implementations that implement AES192 must also implement AES128 and AES256

(I am stating my understanding of the proposal/plan of record here in the issue statement. If there is disagreement, we will modify the issue statement to indicate this.)

AndyGlew, 2020-08-27:
Related issues:

#38 <-- combination of #36 and #37

grnewell, 2020-09-26:
The "Z" naming groups are cumulative, i.e., the union of all those used.  It is easier to define several common groups without AES192, and then define a group with just AES-192 that can be added to any of the former.  Otherwise, the number of named groups explodes... and for AES-192 this would be a waste.

ben-marshall, 2020-10-23:
As decided at meeting 23/oct/2020 - closing all feature-group related issues and tracking in Jira.

closed 2020-10-23 by ben-marshall