riscv/riscv-crypto #38: Crypto subsets: {AES128+AES256}, {AES128+AES256+AES192}

AndyGlew, 2020-08-27:
PROPOSAL / POR

The only permitted constrained subsets of AES are

Both AES128 + AES256 (but not AES192)

All of  AES128 +  AES256 + AES192

Disallowed combinations:

 any combination that does not have both AES128 and AES256

 AES192

---

This issue/ proposal/POR   logically depends on previous issues

AndyGlew, 2020-08-27:
Related issues:

#38 <-- combination of #36 and #37

grnewell, 2020-08-29:
My view is that AES-128 and AES-256 are both essential for any cryptographic instruction set extension (ISE).  So, in order to reduce the number of combinations and permutations of instruction groups it makes sense to lump these together in the vector crypto extension groups.  The area cost of supporting both is not too great, as most of the HW can be used for both.

It should be noted that the scalar approach proposed does not have a similar distinction for AES.  The proposed instruction are needed for all key sizes.  

WRT SHA2, there are some specific instructions for SHA-256 and SHA-512.  In this case I take the opposite position:  SHA-512 is not as essential as SHA-256, and cost of the vector implementation of SHA-512 is much higher, and also drives the vector register bit-lengths up, which might not be as feasible for some smaller vector machines.  So, my recommendation is that SHA-512 be optional.  In groups where SHA-512 is implemented, then SHA-256 should also be.

ben-marshall, 2020-10-23:
As decided at meeting 23/oct/2020 - closing all feature-group related issues and tracking in Jira.

closed 2020-10-23 by ben-marshall