riscv/riscv-crypto #39: Crypto subsets:   should there be a profile that has both all-rounds and by-round  instructions  ?in same

AndyGlew, 2020-08-27:
the crypto instructions come in flavors

1. all-rounds -   a single instruction that performs all of the rounds of the encryption algorithm (like AES,  which has a different number of rounds for AES128, AES192, and AES256)

2. by-round -  instructions that  perform only a single round,  or perhaps 2 or a few rounds, of the  encryption algorithm.  

 there will almost certainly be "profiles" that support only the by-round  instructions.

 there will certainly be profiles that support the  all-rounds instructions.

Q: should there be a profile that supports BOTH the all-rounds and the by-rounds instructions?

Q: should there be a prodile that supports only the all-rounds but not the by-rounds instructions?

I.e.

YES:  by-rounds (but not all-rounds)

YES?: all-rounds but not by-rounds
* this was presented in RN's proposal discussed 8/27)

??: by-rounds PLUS all-rounds

AndyGlew, 2020-08-27:
I (Glew) contend that general purpose machines will want to be able to run software compiled elsewhere.  E.g. if there is working software that runs well enough using the by-rounds instructions, why require it to be recompiled or rewritten to use the all-rounds instructions?

Embedded systems might not care ... if every piece of SW is recompiled or rewritten for the current target.  But actually many embedded systems receive opaque binaries from code vendors, that the recipient cannot recompile.

 folks might say that all of these crypto instructions will be placed in libraries that can be re-linked. But of course that's not always true. Shared libraries in general are a security vulnerability if not managed  appropriately;  even relinked both static libraries can cause issues.    from software vendors ship  statically linked, not relinkable, code to  reduce support costs.

---

The logical extension of this might be to have  or allow processors that implement all possible crypto instructions;  indeed, all possible RISC-V instructions.   while this is true in a hypothetical or academic sense, is not necessarily practically true:  we only care about running different flavors of the same instruction when it is commercially important, e.g. when software vendors may ship opaque binaries and one flavor and not the other.  

 Software vendors will often ship a single code that runs on a lowest common denominator system,  if the performance is acceptable. Software vendors only want to be dragged into supporting advanced instructions like the "all-rounds"  crypto instructions if they really need the performance.  some crypto use cases need to performance, some don't.  

---

I therefore contend that a profile that supports BPTH all-rounds and by-round Crypto instructions should be supported.

I am agnostic as to whether a profile that allows all-rounds without by-round should be supported. Except that eliminating supported combinations is always good.

grnewell, 2020-08-29:
Just a reminder that the main incentive for the all-rounds crypto instruction set is so that implementers can create side-channel analysis (SCA) -resistant implementations using common all-digital countermeasures such random masking.  The per-round instructions, by definition, require writing intermediate results to the vector registers, making prevention of leakage of these intermediates next to impossible.  

Performance is a secondary benefit.  Completing all the rounds in HW vs in SW does open opportunities for better performance and energy consumption, at the expense (typically) of greater latency for the all-rounds instructions compared to instructions that only do one round.  Even so, the latency does not have to be much, if any, greater than some other RISC-V vector instructions.

grnewell, 2020-09-26:
The instruction groups can be merged by listing more than one group using the "Z" naming convention.  Therefore, it is possible to have a group with per-round and another group with all-rounds.  Then, the way to name the union is just to list both groups.
I have proposed standard groups with per-round or all-round instructions exclusively, as I think in most cases if one is implemented the other will not be.  However, it is possible for the implementer to create the union of these, if they so wish.

ben-marshall, 2020-10-23:
As decided at meeting 23/oct/2020 - closing all feature-group related issues and tracking in Jira.

closed 2020-10-23 by ben-marshall