riscv/riscv-crypto #40: Work with BitManip TG to ensure they have groupings that work forCrypto

grnewell, 2020-08-27:
Crypto TG needs to work with BitManip TG on instruction groupings for the "naming/branding" strings.  For us (Crypto) it is important that the BitManip groupings can be used to make sure that the instructions we are "borrowing" from them can be succintly express in their naming strings.  We also need to make sure that their CLMUL instruction is constant-time (i.e., latency is data independent, as a minimum leakage-related requirement).

ben-marshall, 2020-08-28:
We should certainly keep up communication with Bitmanip, but - would it be easier to just define a "feature string" which refers to "All of the Bitmanip instructions which are used by crypto"? That way we only need to worry about picking a unique name for it, rather than having to depend on them for anything. I'd assumed that all of the Bitmanip-borrowed instructions would just come as a single group, but am happy to be convinced otherwise.

About the constant time aspect of CLMUL - This is already expressed in the draft crypto spec. Perhaps asking Claire to note this as an issue would be sensible?

grnewell, 2020-08-29:
I suggested this approach (i.e., going it alone) to Mark and indicated he would rather BitManip. TG provide all the groupings of their instructions... even those groupings we see as essential.

ben-marshall, 2020-09-01:
Hmm. I don't think that's the best way round IMHO. Better to let Bitmanip define the most sensible groupings *for Bitmanip* and let us do the same for Crypto. Otherwise you get unnecessary weird inter-dependencies between TGs.

By borrowing instructions from Bitmanip, we've already broken the notion that there is a 1-to-1 correspondence between an instruction and the extension it belongs too. Since sharing instructions between extensions is sensible, forcing ownership of issues around those instructions on only the TG which originally defined it doesn't make sense like it did before.

I may be preaching to the choir now.

grnewell, 2020-10-22:
We (i.e., Crypto TG) are now taking the reins an proposing a named instruction group ("Zkb") for the bit-manipulation instructions we borrow from the Bit Manipulation TG.  A JIRA ticket has been filed as an "FYI" to the Bit-Manip TG.

ben-marshall, 2020-10-23:
As decided at meeting 23/oct/2020 - closing all feature-group related issues and tracking in Jira.

closed 2020-10-23 by ben-marshall