riscv/riscv-crypto #42: Use of "dots" in assembly instruction names

grnewell, 2020-08-29:
Issue to track how dots are used in the proposed instruction assembly names.

grnewell, 2020-08-29:
I would remove the “.” unless the two (or more) instructions are really very close variations of one operation.  The .vv, .vi, .vs, .vx variations of essentially the same vector instruction are an example of this.  If the instructions represent different basic operations, then I would refrain from using the base + dot + suffix format to name them.  

Therefore, I would propose removing the dots from all the current mnemonics such as  saes32.encs, saes32.encsm where the underlying computation is different.  (Note, I also recommend removing the leading "s")

ben-marshall, 2020-09-01:
> Therefore, I would propose removing the dots from all the current mnemonics such as saes32.encs, saes32.encsm where the underlying computation is different. (Note, I also recommend removing the leading "s")

Agreed with both points. I'll take this action once we know exactly what each instruction *should* be called.

commit 23d35923 (2020-09-22): Implement instruction re-naming.

- Closes #48 - Instruction Renaming
- Closes #46 - Shorter names, and consistency for instructions with immediate operands
- Closes #45 - Vector instruction names and dot-extensions
- Closes #42 - Use of "dots" in assembly instruction names
- Closes #41 - Change scalar instruction names to remove 's' prefix.

This is a squash commit. It contains all of the commits made on
the `dev/instruction-renaming` branch.

---

commit 93be8b084e44fe2918264f967eb339a7526c15c9
Author: Ben Marshall <ben.marshall@bristol.ac.uk>
Date:   Tue Sep 22 13:49:29 2020 +0100

    sail [... cut]
closed 2020-09-22 by ben-marshall