riscv/riscv-crypto #44: xperm.* variations.

ben-marshall, 2020-09-03:
Context:
- Originally, the scalar crypto spec included a `lut4` instruction, which did nibble-wise lookups on values in registers and is very useful for small SBoxes or permutations.
- This instruction was deemed more generally useful than Crypto, so the Bitmanip TG took ownership of it.
  - They changed the name to `xperm`, and added byte and halfword variants as well.
- When I updated the crypto spec to point at the Bitmanip spec, I accidentally included the extra byte and halfword variants without thinking.

Action:
- I've removed the extra variants from the spec on the `dev/next-release` branch for now.
- If we decide to keep them, I'll add them back to the spec. Otherwise they can be left out. This leaves the spec as it should have been when the Bitmanip TG first took ownership of the instruction.

grnewell, 2020-09-04:
If I understand correctly, the byte variant would be helpful in computing a LUT8.  I can see this being useful for crypto., esp. when computing masked 8-bit S-boxes for the purpose of DPA hardening.

ben-marshall, 2020-09-04:
Interestingly, Claire put together an example where you only need the 4-bit (`xperm.n`) variant to do an arbitrary 8-bit SBox operation, which makes me think that the wider variants might not be needed.

[See here](http://svn.clairexen.net/handicraft/2020/lut4perm/).

Looking at that again though, I realise that we might have a glaring omission: the `gorc` instruction is really useful in that example, and is not currently included in our spec.

I'll probably raise this in the next meeting or put an email out in the meanwhile. Happily, `gorc` can be implemented very cheaply alongside `grev` which we already have.

ben-marshall, 2020-09-11:
Decided at meeting on 10'th Sept 2020 - `xperm.b` and `xperm.n` will be included in the scalar crypto spec.

commit 555afd7c (2020-09-11): Fix #44 - xperm.* Variations

- Decided at meeting September 10'th, 2020

- Include xperm.n and xperm.b from Bitmanip.

 On branch dev/next-release
 Your branch is ahead of 'origin/dev/next-release' by 1 commit.
   (use "git push" to publish your local commits)

 Changes to be committed:
	modified:   doc/riscv-crypto-spec.bib
	modified:   doc/tex/sec-scalar-bitmanip.tex

 Changes not staged for commit:
	modified:   extern/riscv-gnu-toolchain (modified content)
	modified:   extern/riscv-isa-sim (modified content)
closed 2020-09-22 by ben-marshall