riscv/riscv-crypto #48: Instruction Renaming

ben-marshall, 2020-09-11:
This issue pulls together several others:
- #46 Shorter names, and consistency for instructions with immediate operand
- #45 Vector instruction names and dot extensions
- #42 Use of dots in assembly instruction names
- #41 Change scalar instruction names to remove `s` prefix.

The table below lists the *current* mnemonic and the *proposed* new mnemonic. Please consider each proposal, and if you have any thoughts or suggestions for better alternatives, please discuss in the comments below.

This issue will be left alive for a week to gather feedback before being closed and the decided changes implemented.

**Scalar Instructions:**

Principles:

- No `.` characters.
- No leading `s` for the scalar instructions.
- Use shorter mnemonics where possible.
- Suffix with an `i` if the instruction contains an immediate.

Current mnemonic | Proposed mnemonic | Notes
----------|-------------------|-----------------
`saes32.encs`   | `aes32esi `   | AES RV32 encrypt: sub-bytes, immediate
`saes32.encsm`  | `aes32esmi`   | AES RV32 encrypt: sub-bytes + mix-columns, immediate
`saes32.decs`   | `aes32dsi`    | AES RV32 decrypt: sub-bytes, immediate
`saes32.decsm`  | `aes32dsmi`   | AES RV32 decrypt: sub-bytes + mix-columns, immediate
`saes64.ks1`    | `aes64ks1i`   | AES RV64 key-schedule "one" [sub-bytes, rotate, immediate round-constant]
`saes64.ks2`    | `aes64ks2`    | AES RV64 key-schedule "two" [XOR summation]
`saes64.imix`   | `aes64im`     | AES RV64 inverse-mix-columns
`saes64.encsm`  | `aes64esm`    | AES RV64 encrypt shift-rows, sub-bytes, mix-columns
`saes64.encs`   | `aes64es`     | AES RV64 encrypt shift-rows, sub-bytes
`saes64.decsm`  | `aes64dsm`    | AES RV64 decrypt shift-rows, sub-bytes, mix-columns
`saes64.decs`   | `aes64ds`     | AES RV64 decrypt shift-rows, sub-bytes
`ssha256.sig0`  | `sha256sig0`  | SHA256 RV32/64 "sigma-zero" function
`ssha256.sig1`  | `sha256sig1`  | SHA256 RV32/64 "sigma-one" function
`ssha256.sum0`  | `sha256sum0`  | SHA256 RV32/64 "sum-zero" function
`ssha256.sum1`  | `sha256sum0`  | SHA256 RV32/64 "sum-one" function
`ssha512.sum0r` | `sha512sum0r` | SHA512 RV32 "sum-zero" function w/ rotate
`ssha512.sum1r` | `sha512sum0r` | SHA512 RV32 "sum-one" function w/ rotate
`ssha512.sig0l` | `sha512sig0l` | SHA512 RV32 "sigma-zero" function, low
`ssha512.sig0h` | `sha512sig0h` | SHA512 RV32 "sigma-zero" function, high
`ssha512.sig1l` | `sha512sig1l` | SHA512 RV32 "sigma-one" function, low
`ssha512.sig1h` | `sha512sig1h` | SHA512 RV32 "sigma-one" function, high
`ssha512.sum0`  | `sha512sum0r` | SHA512 RV64 "sum-zero" function
`ssha512.sum1`  | `sha512sum0r` | SHA512 RV64 "sum-one" function
`ssha512.sig0`  | `sha512sig0h` | SHA512 RV64 "sigma-zero" function
`ssha512.sig1`  | `sha512sig1l` | SHA512 RV64 "sigma-one" function
`ssm3.p0`       | `sm3p0`       | SM3 RV32/64 "p-zero" function
`ssm3.p1`       | `sm3p1`       | SM3 RV32/64 "p-one" function
`ssm4.ed`       | `sm4ed`       | SM4 RV32/64 encrypt-decrypt
`ssm4.ks`       | `sm4ks`       | SM4 RV32/64 key-schedule
`pollentropy`   | `pollentropyi`| RV32/64 entropy source.

**Vector Instructions**

Principles:

- `.` characters used only to differentiate between source operand specifiers, in line with base
  vector spec.
- Make sure `.v*` suffixes are consistent with base vector spec.
- Suffix with an `i` if the instruction contains an immediate.

Current mnemonic | Proposed mnemonic | Notes
----------|-------------------|-----------------
`vaese.vs       ` | `vaese.vs         ` | AES Single Round Encrypt Vector-Scalar (First N rounds)
`vaeselast.vs   ` | `vaeselast.vs     ` | AES Single Round Encrypt Vector-Scalar (Final Round)
`vaesd.vs       ` | `vaesd.vs         ` | AES Single Round Decrypt Vector-Scalar (First N rounds)
`vaesdlast.vs   ` | `vaesdlast.vs     ` | AES Single Round Decrypt Vector-Scalar (Final Round)
`vaese.vv       ` | `vaese.vv         ` | AES Single Round Encrypt Vector-Vector (First N rounds)
`vaeselast.vv   ` | `vaeselast.vv     ` | AES Single Round Encrypt Vector-Vector (Final Round)
`vaesd.vv       ` | `vaesd.vv         ` | AES Single Round Decrypt Vector-Vector (First N rounds)
`vaesdlast.vv   ` | `vaesdlast.vv     ` | AES Single Round Decrypt Vector-Vector (Final Round)
`vaes128keyi    ` | `vaes128key.vi    ` | AES 128 Single Round key schedule step (Forward/Encrypt)
`vaes128invkeyi ` | `vaes128invkey.vi ` | AES 128 Single Round key schedule step (Inverse/Decrypt)
`vaes192keyi    ` | `vaes192key.vi    ` | AES 192 Single Round key schedule step (Forward/Encrypt)
`vaes192invkeyi ` | `vaes192invkey.vi ` | AES 192 Single Round key schedule step (Inverse/Decrypt)
`vaes256keyi    ` | `vaes256key.vi    ` | AES 256 Single Round key schedule step (Forward/Encrypt)
`vaes256invkeyi ` | `vaes256invkey.vi ` | AES 256 Single Round key schedule step (Inverse/Decrypt)
`vaese128.vs    ` | `vaese128.vs      ` | AES 128 All Rounds Encrypt Vector-Scalar
`vaese192.vs    ` | `vaese192.vs      ` | AES 192 All Rounds Encrypt Vector-Scalar
`vaese256.vs    ` | `vaese256.vs      ` | AES 256 All Rounds Encrypt Vector-Scalar
`vaesd128.vs    ` | `vaesd128.vs      ` | AES 128 All Rounds Decrypt Vector-Scalar
`vaesd192.vs    ` | `vaesd192.vs      ` | AES 192 All Rounds Decrypt Vector-Scalar
`vaesd256.vs    ` | `vaesd256.vs      ` | AES 256 All Rounds Decrypt Vector-Scalar
`vaese128.vv    ` | `vaese128.vv      ` | AES 128 All Rounds Encrypt Vector-Vector
`vaese192.vv    ` | `vaese192.vv      ` | AES 192 All Rounds Encrypt Vector-Vector
`vaese256.vv    ` | `vaese256.vv      ` | AES 256 All Rounds Encrypt Vector-Vector
`vaesd128.vv    ` | `vaesd128.vv      ` | AES 128 All Rounds Decrypt Vector-Vector
`vaesd192.vv    ` | `vaesd192.vv      ` | AES 192 All Rounds Decrypt Vector-Vector
`vaesd256.vv    ` | `vaesd256.vv      ` | AES 256 All Rounds Decrypt Vector-Vector
`vaes128rkey    ` | `vaes128rkey      ` | AES 128 Final round key from first round key
`vaes192rkey    ` | `vaes192rkey      ` | AES 192 Final round key from first round key
`vaes256rkey    ` | `vaes256rkey      ` | AES 256 Final round key from first round key
`vsha2.ws       ` | `vsha2wsi         ` | SHA2  16 Rounds Working State Update
`vsha2.ms       ` | `vsha2ms          ` | SHA2  16 Rounds Message State Update
`vsha2.hs       ` | `vsha2hs          ` | SHA2 All Rounds update hash state.
`vclmul.vv      ` | `vclmul.vv        ` | Vector Carry-less Multiply Vector-Vector
`vclmul.vs      ` | `vclmul.vs        ` | Vector Carry-less Multiply Vector-Scalar
`vclmulh.vv     ` | `vclmulh.vv       ` | Vector Carry-less Multiply High Vector-Vector
`vclmulh.vs     ` | `vclmulh.vs       ` | Vector Carry-less Multiply High Vector-Scalar
`vclmacc.vv     ` | `vclmacc.vv       ` | Vector Carry-less Multiply Accumulate Vector-Vector
`vclmacc.vs     ` | `vclmacc.vs       ` | Vector Carry-less Multiply Accumulate Vector-Scalar
`vclmacch.vv    ` | `vclmacch.vv      ` | Vector Carry-less Multiply Accumulate High Vector-Vector
`vclmacch.vs    ` | `vclmacch.vs      ` | Vector Carry-less Multiply Accumulate High Vector-Scalar
`vwclmul.vv     ` | `vwclmul.vv       ` | Vector Widening Carry-less Multiply Vector-Vector
`vwclmul.vs     ` | `vwclmul.vs       ` | Vector Widening Carry-less Multiply Vector-Scalar
`vwclmacc.vv    ` | `vwclmacc.vv      ` | Vector Widening Carry-less Multiply Accumulate Vector-Vector
`vwclmacc.vs    ` | `vwclmacc.vs      ` | Vector Widening Carry-less Multiply Accumulate Vector-Scalar
`vrot.vv        ` | `vrot.vv          ` | Vector Rotate - Vector-Vector
`vrot.vs        ` | `vrot.vs          ` | Vector Rotate - Vector-Scalar
`vrot.vi        ` | `vrot.vi          ` | Vector Rotate - Vector-Immediate
`vrot.vx        ` | `vrot.vx          ` | Vector Rotate - Vector-XReg
`vgrev.vv       ` | `vgrev.vv         ` | Vector GREV - Vector-Vector
`vgrev.vs       ` | `vgrev.vs         ` | Vector GREV - Vector-Scalar
`vgrev.vi       ` | `vgrev.vi         ` | Vector GREV - Vector-Immediate
`vgrev.vx       ` | `vgrev.vx         ` | Vector GREV - Vector-XReg

JamesKenneyImperas, 2020-09-12:
I think a few of the scalar names could be incorrect. For example, should ssha512.sum1r should be renamed sha512sum**1**r, not sha512sum0r?

ben-marshall, 2020-09-12:
@JamesKenneyImperas - Gah, well spotted. Copy pasta on my part. I've updated the table. Thanks a lot!

grnewell, 2020-09-14:
We should define all the vector commands with the appropriate "dot" extension(s):  one  or more of ".vv", ".vi", ".vx", ".vs" even if the assembler folks decide that dropping some of them in the assembly mnemonics is allowed.  There are several in the table where the extension is missing.

ben-marshall, 2020-09-15:
@grnewell  - you mean `vsha2wsi`, `vsha2ms`, and `vsha2hs`?

Which should they be? I can't remember seeing anything on this in the summit slides I derived the current specs from.

grnewell, 2020-09-15:
Also, besides those SHA instructions, the extension is missing on the reverse-key instructions in the table above.

In the case of the SHA instructions, only .vv really makes any sense.
For the reverse-key instructions we had decided that we would only support .vv.  If the programmer wants to do only one element they can set vl=1.

ben-marshall, 2020-09-16:
@grnewell - I remember now. Thanks. I've updated the table accordingly. :+1:

commit 23d35923 (2020-09-22): Implement instruction re-naming.

- Closes #48 - Instruction Renaming
- Closes #46 - Shorter names, and consistency for instructions with immediate operands
- Closes #45 - Vector instruction names and dot-extensions
- Closes #42 - Use of "dots" in assembly instruction names
- Closes #41 - Change scalar instruction names to remove 's' prefix.

This is a squash commit. It contains all of the commits made on
the `dev/instruction-renaming` branch.

---

commit 93be8b084e44fe2918264f967eb339a7526c15c9
Author: Ben Marshall <ben.marshall@bristol.ac.uk>
Date:   Tue Sep 22 13:49:29 2020 +0100

    sail [... cut]
closed 2020-09-22 by ben-marshall