riscv/riscv-crypto #50: Decode conflict (pollentropy and sha256sig0)

JamesKenneyImperas, 2020-09-18:
Although the instruction encodings are only tentative, it looks to me like there is a decode conflict between the pollentropy instruction (with shamtw=0) and sha256sig0:

```
0000111 00000  rs1   111 rd 0101011 ssha256.sig0
0000111 shamtw 01010 111 rd 0101011 pollentropy
```

(When shamtw!=0, there are conflicts with other instructions as well.)

ben-marshall, 2020-09-18:
Hi James
You're absolutely right, I'm sorry I should have noted this in an issue. This has been fixed on the `dev/next-release` branch in [this](https://github.com/riscv/riscv-crypto/commit/bf86ef1af2fd2eb60af43e7fb3ec2efd61f9ed73) commit, but not merged into `main` yet. It should be merged in the next 7 days, I've been waiting until any instruction name changes settled down before merging it. I'll close this issue when everything gets merged.
Thanks,
Ben

JamesKenneyImperas, 2020-09-18:
Hi Ben,

Thanks - I can see the correct decode now, so I'll use that.

James

ben-marshall, 2020-09-22:
Hi @JamesKenneyImperas 
The relevant commit has been merged into master now. If you're alright to close this, that'd be great. Thanks for pointing it out.
Cheers,
Ben

JamesKenneyImperas, 2020-09-22:
Now fixed - closing.

closed 2020-09-22 by JamesKenneyImperas