riscv/riscv-crypto #51: Spike K implementation

JamesKenneyImperas, 2020-09-30:
I'm using Spike with the patch given here (taken on 28th September) and I've got some feedback and questions.

1. When running with RV32, most instructions work as I expect (although pollentropy seems to be a nop - maybe generating a deterministic valid sequence for test purposes might be more useful?)

2. When running with RV64, I'm seeing a few unexpected things:
2.1. All four `saes32` instructions seem to execute - I would have expected them to be Illegal Instructions according to the specification. In addition, they seem to update only the low half of the destination register, leaving the upper half unchanged.
2.2. All six `sha512` instructions that are specified as RV32-only also execute, zero extending the result -  - I would have expected them to be Illegal Instructions according to the specification.
2.3. The two `ssm3` instructions seem to be producing an unexpected 64-bit result. Looking at the specification, I would have expected these to produce a 32-bit zero-extended result.

Can you please confirm what the correct behavior should be for these instructions?

Thanks.

JamesKenneyImperas, 2020-09-30:
A correction - the `saes32` instructions seem to update the high half of the result with a copy of the high half of the second argument register.

ben-marshall, 2020-09-30:
Hi @JamesKenneyImperas 

1. Pollentropy is indeed a `nop` right now. The idea was to discuss the best way to model it with the TG. I opened a dedicated issue for this (#28), because whatever methodology we pick should be followed across Spike, QEMU, OVPSim and friends. Clearly that fell by the wayside and I'll try and pick that up again.

2. Yes, there is currently no checking anywhere for whether or not the K extension instructions (of any flavour) should generate an Illegal Opcode Exception. I'd left this out until we had finished discussing the feature groups / instruction subsets. Clearly the RV32/RV64 instructions can be safely specified though. Generally, what you were expecting given the spec is what should happen:

    2.1. The `aes32` instructions *must* raise an illegal instruction exception when executed on RV64. In the same way that the `aes64` instructions *must* raise an illegal instruction exception when executed on RV32.

    2.2. Likewise, the RV32 `sha512` instructions *must* raise an illegal instruction exception on RV64.

    2.3. The `sm3`  instructions should execute on RV64, and like you say, produce a zero extended 32-bit result.

I hope that's enough for you to work with? Please let me know if not. I'll make the adjustments to Spike, and let you know when they're finished so we can check everything is in agreement?

Note that the checks are also mostly absent from the SAIL model too.

Cheers,
Ben

JamesKenneyImperas, 2020-09-30:
Hi Ben,

That all makes sense - I'll check everything again when you let me know that there's a new patch.

Thanks!

James.

ben-marshall, 2020-09-30:
Hi @JamesKenneyImperas 

If you have a look on the `dev/issue/51` branch, commit 42d8da2 should have fixes for the AES, SHA512 and SM3 issues:

>- All aes32* instructions now have the "require_rv32" macro, meaning they will raise
  an "Illegal Opcode" exception if executed on RV64. This is as the spec says.

> - Likewise for the RV32 only sha512* instructions.

> - Added explicit zext_xlen(*) to the SM3 and SM4 instruction results, since their
  results are always 32-bit, but they can be executed on RV32 and RV64.

If that all meets your expectation, I'll merge that into `dev/next-release` and `master` and close this issue.

I'll ask on the mailing list about the `pollentropy` approach later today.

Cheers,
Ben

JamesKenneyImperas, 2020-10-01:
Hi @ben-marshall,

I've looked at the branch - all issues seem corrected except for `sm3p0` and `sm3p1`, which still look like they might be wrong on RV64. I first ran this example on RV32:

```
        li          s0, 0x5de973ac
        sm3p0       s1, s0               // s1 -> 0x685690c5
        sm3p1       s1, s0               // s1 -> 0x3211a9e1
```

This matches the `riscvOVPsim` implementation I've done. I then ran this example on RV64:

```
        li          s0, 0x9f5417f65de973ac
        sm3p0       s1, s0               // s1 -> 0x189645b348fa34c5
        sm3p1       s1, s0               // s1 -> 0x6fc086b31938a9e1
```

So I'm still seeing 64-bit non-zero-extended results. `riscvOVPsim` reckons the answers should be:

```
        sm3p0       s1, s0               // s1 -> 0x00000000685690c5
        sm3p1       s1, s0               // s1 -> 0x000000003211a9e1
```

(i.e. zero-extended equivalents of the RV32 answers, since the bottom 32 bits of s0 are identical for both runs.)

It's quite possible I've misunderstood the specification or that we've messed up applying the patch - if so can you let me know?

Thanks,

James.

ben-marshall, 2020-10-01:
Hi @JamesKenneyImperas 

You're right, I'd forgotten that Spike treats the `RS1` and `RS2` macros as *always* 64-bit values, whether running in RV32 or RV64 mode, and it all went wrong from there.

I've updated the patch (See c8b9356 on `dev/issue/51`) and now get the same results as you for `sm3p0` and `sm3p1` for the same inputs.

Let me know if it still doesn't match your expectation. I'll work on adding a proper SM3 algorithm test to the repository too, but that might take a little longer.

Cheers,
Ben

JamesKenneyImperas, 2020-10-01:
Hi Ben,

Ok, I will check again. I think it is likely that we will now match for both RV32 and RV64. I'll let you know!

James.

JamesKenneyImperas, 2020-10-02:
Hi @ben-marshall,

With that last patch, Spike and riscvOVPsim now match, as expected. Do you want me to close this issue now?

James.

ben-marshall, 2020-10-02:
Hi @JamesKenneyImperas 

That's great. If you leave the issue open for now, I'll close it when it gets merged into the `master` branch.

Cheers, and thanks for your help/patience!
Ben

commit 4cd93ead (2020-10-02): Merge branch 'dev/issue/51' into dev/next-release

- Fixes #51 - Spike K implementation - some issues with consistency across RV32/64

 Date:      Fri Oct 2 15:27:42 2020 +0100

 On branch dev/next-release
 Your branch is ahead of 'origin/dev/next-release' by 5 commits.
   (use "git push" to publish your local commits)

 Changes to be committed:
	modified:   tools/patch-spike.patch

 Changes not staged for commit:
	modified:   extern/riscv-gnu-toolchain (modified content)
	modified:   extern/riscv-isa-sim (modified content)
closed 2020-10-08 by ben-marshall