riscv/riscv-crypto #52: Crypto subsets:  Should their be scalar crypto intruction groups w/o pollentropy?

grnewell, 2020-10-02:
Currently there are only three top-level groups of scalar crypto instructions:  1)  The "NIST Suite" in the RV32 context, 2) the same in the RV64 context, and 3)  the SM3/SM4 instruction group.

Both the RV32 and RV64 NIST Suites include the _pollentropy_ instruction.

The concern is that some CPU/SoC implementers may not want to implement their random source using _pollentropy_.  Building the remainder of the NIST Suite without _pollentropy_ from the lowest level primitives will result in long and cumbersome names.  Should we add named groups similar to the NIST Suites, but without _pollentropy_, so that this combination of commands can be expressed concisely?

JamesKenneyImperas, 2020-10-02:
On a vaguely related note, why is entropy found using a special instruction instead of having a read-only `entropy` CSR instead, in a similar manner as (for example) `cycle` and `instret`?

A CSR would seem to me to be a more natural fit for this kind of thing.

ben-marshall, 2020-10-03:
> why is entropy found using a special instruction instead of having a read-only entropy CSR instead

We did consider this. I think the rationale for siding with an instruction (which we/I should definitely do a better job of documenting) was along the lines of:

1. There are many many more instruction encodings available than CSR encodings. Particularly M-mode, read-only CSRs, which is what `pollentropy` would be. Read only CSRs are even more wasteful, because they effectively turn all of the CSR instruction encodings with write behaviour into `nop`s.

2. Having a dedicated instruction allows for *somewhat* more flexibility in the implementation, and implies nothing about the actual entropy source being per-HART or anything like that. CSRs have a certain amount of baggage with that sort of thing.

Ultimately, it could very reasonably have been a CSR, but defining it as its own instruction made specifying it easier, as it has no cross cutting behaviour with the rest of the privileged specification that needs considering.

JamesKenneyImperas, 2020-10-03:
Hi Ben,

I don't think CSR address limitations should be of concern: of the 192 available standard MRO CSR addresses, I think only 4 are taken currently (mvendorid, marchid, mimpid and mhartid) so there should be plenty of space left. I can't comment on whether the instruction being declared in CSR space necessarily implies it is per hart, or on the difficulties involved in getting CSR numbers assigned - I'm sure you know more about this than I do.

James.

JamesKenneyImperas, 2020-10-21:
Hi Ben,

I haven't monitored the precise details of the recent changes here, but did I see that pollentropy has now been reclassified as a CSR access? If so, is there any chance you could generate new version of the pdf specification that reflects this?

Thanks,

James.

ben-marshall, 2020-10-21:
Hi James

Yes, I'm planning to do another spec release in the next 5 working days or so. The pollentropy/CSR change hasn't been finalised yet (hence no new PDF), but we have the regular meeting on Thursday evening, so hopefully it will all be sorted then. I'll comment here as well when I do another release.

Cheers,
Ben

ben-marshall, 2020-10-23:
@JamesKenneyImperas - The new pollentropy spec has been merged. I'll do a release + PDF shortly.
Cheers,
Ben

ben-marshall, 2020-10-23:
As decided at meeting 23/oct/2020 - closing all feature-group related issues and tracking in Jira.

closed 2020-10-23 by ben-marshall