riscv/riscv-crypto #56: RV32 build issue

pmundkur, 2020-10-13:
I'm trying to build your RV32 model in an attempt to reproduce #54. I did 
```
$ export RISCV_ARCH=riscv32-unknown-elf
$ source bin/conf.sh 
$ ./tools/start-from-scratch.sh 
```
Perhaps this is not quite right? However, the toolchain build terminates when trying to build `riscv-pk` with
```
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
gcc: error: unrecognized argument in option ‘-mabi=ilp32’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
```
I've had such problems building RV32 toolchains for some time now, resulting in the Sail RV32 model not being well tested recently.

ben-marshall, 2020-10-13:
Hi there, 

I think you need to set `RISCV_ARCH=riscv64-unknown-elf`? That's how it is described in the `tools/README.md` file. I should probably make it clearer that doing it that way gives you a toolchain which can build *both* RV32 and RV64.

> I've had such problems building RV32 toolchains for some time no

You and me both. The think that made it easier for me was to just stop building RV32-only toolchains.

I hope that helps, let me know if not and I'll try and reproduce whatever problem you have locally. I won't reply for a bit though, since it's late in my timezone now. Best of luck!

Cheers,
Ben

ben-marshall, 2020-10-13:
Sorry, one more thought - if you still can't get it to work, can you attach the output of:

- `tools/pk-conf.sh`
- `tools/pk-build.sh`

Since they will have all the information about the PK build environment. It looks like the error you are getting implies the wrong GCC is being used (i.e. the host system x86 compiler, rather than the newly built RISC-V one).

pmundkur, 2020-10-14:
Thanks, I got a working build and can repro the issue.

closed 2020-10-14 by pmundkur

ben-marshall, 2020-10-14:
Great, glad to hear it :+1: