riscv/riscv-crypto #63: Removing reg-reg-reg variants of permutation instructions.

ben-marshall, 2020-11-06:
A/O the meeting on Nov 6'th 2020, we have decided to remove the reg-reg-reg variants of:
- `gorc`
- `grev`
- `shufl`
- `unshufl`

Because we do not have a use case for them.

We are keeping the reg-reg-reg variants of
- `ror`
- `rol`

Because we anticipate there *may* be a use case for them in the future, and they are extremely cheap to implement given the presence of the immediate variants.

JamesKenneyImperas, 2020-11-06:
Morning Ben,

A couple of clarifications:
1. When you say the instructions are _removed_, I assume you mean only from the K extension, not the B extension. I'm pretty sure this is the case, but just checking.
2. On the spreadsheet you just updated, I notice that `gorciw` is also marked as not present, but this wasn't mentioned in the list above. Is it also removed?

Thanks,

James.

JamesKenneyImperas, 2020-11-06:
Ah no, I think I misread the spreadsheet. It's just not in a feature subset.

JamesKenneyImperas, 2020-11-06:
Correction, it does seem to be removed. Aaargh.

ben-marshall, 2020-11-06:
Hi James

1. Yes, removed from the K extension. The B extension remains unchanged. I'll update my issue description above to clarify.
2. I will double check with @mjosaarinen about `gorciw` it is *currently* included in the scalar K spec, I think that the spreadsheet might just have a typo.

Broadly, we are removing the 3-reg address versions of `gorc`, `grev`, `shufl`, `unshufl`, but are keeping all of the immediate variants `gorci`, `grevi`, `shufli`, and `unshufli`.

mjosaarinen, 2020-11-06:
Hi,

Talked about this with Ben.. I think we really only need a subset of these immediate variants (rev.b, rev8, zip, unzip) .. apologies for the confusion, I'm to blame for some of the things in that spreadsheet.

Cheers,
- markku

ben-marshall, 2020-11-12:
@JamesKenneyImperas - the new `v0.7.2-scalar` specification release should detail all of this explicitly. I'll leave this issue open for now in case there are other questions?

JamesKenneyImperas, 2020-11-16:
Hi Ben,

I think this is now clear to me - thanks.

James.

ben-marshall, 2020-11-16:
Hi James
Glad to hear it, I'll close this for now then.
Cheers,
Ben

closed 2020-11-16 by ben-marshall