riscv/riscv-crypto #65: Decode for sm4 and aes32 instructions, rs1==rd

JamesKenneyImperas, 2020-12-04:
Hi Ben,

I notice that these instructions are specified to require rs1==rd, with the intention that the number of encoding points is reduced.

I wonder whether it would be better just to require the rs1 field to be a specified constant (probably 0) instead? This would free up the same number of encoding points in a more useful manner.

James.

ben-marshall, 2020-12-04:
Hi James

So, yes. We expected to have this discussion. The rationale for `rs1==rd` rather that setting `rs1=0` and having to source it from the `rd` part of the instruction is to avoid having an extra multiplexor on the decode path for register addresses. RISC-V is proud of (bar the C extension) always having register addresses in the same place, and we didn't want to break that.

That said, I think your suggestion is a perfectly good one. We could have gone that way, but didn't. Hopefully this release will trigger some wider discussion and consensus. If we did go down the route you suggest, I'd probably advocate for setting `rd=0` in the encoding, and then sourcing the destination register from `rs1`, since it's `rs1` we need to decode very quickly.

Cheers,
Ben

JamesKenneyImperas, 2020-12-04:
Hi Ben,

Ok, I'll leave the discussion to the experts, since I'm far removed from hardware. All I'll say is that the proposal as is prevents all the unused encoding points being used by any instruction that requires a GPR in either field, so the freed points won't be useful for much.

James.

rdolbeau, 2020-12-07:
Hello all,

I updated my support of Zkn[eh] for [VexRiscv](https://github.com/SpinalHDL/VexRiscv) (can be found in [VexRiscvBPluginGenerator](https://github.com/rdolbeau/VexRiscvBPluginGenerator/)) to track 0.8, along with the associated test codes.

I couldn't figure out how to implement the rs1 == rd restriction of aes32es[m]i, and ended up not implementing it and accepting all rs1, like in previous drafts. Worked like a charm. Like @JamesKenneyImperas I don't think the 'saved' opcode space can easily be reused anyway... (beware, the test codes won't respect it either so may need modifications to run on a 0.8-compliant implementation).

Opcode regularity is easier to deal with than special-casing register access or matching patterns... Even though in most use cases of those instructions one of the register won't be reused and could be overwritten, it's still worth it to make them behave like regular 2-operands instructions for the sake of simplicity I think.

Cordially,

ben-marshall, 2020-12-07:
Hi all
I've raised this topic on the mailing list (viewable publicly [here](https://lists.riscv.org/g/tech-crypto-ext/topic/decode_for_sm4_and_aes32/78782954)). Please speak up if you think I've missed anything!
Thanks,
Ben

JamesKenneyImperas, 2020-12-07:
As another data point, note that there are instructions in other extensions that use a field to specify both a source and destination register - for example, from the Vector specification:

`vamoaddei8.v vd, (rs1), vs2, vd,  v0.t # Write original value to register, wd=1
`

rdolbeau, 2020-12-10:
Hello,

Looking at the discussion in the mailing list...

a) data-independent timing is not optional for things like AES; it's mandatory for any sane design that could ever be attacked;

b) the relevant instructions are for RV32 only;

c) such instructions are probably the cheapest way to get data-independent timing for AES (vs. some dedicated AES block out-of-core).

So it's likely they end up being implemented in really small RV32 cores... where decoding irregularity is the costliest, I would guess.

To save encoding while avoiding a complex implementation, maybe aes32esmi/aes32esi could be a normal two-operands instruction, and extract 2-3 bits of semantic (2 for BS and perhaps 1 for aes32esmi vs. aes32esi) from the register numbers rather than elsewhere in the encoding space? Though it would push some of the complexity into the compiler... (not a HW guy so I'm not really sure it would be easier to do than using rs1-for-rd, but it saves 3/4 or 7/8 of the encoding space without relying on some non-standard encoding pattern).

Cordially,

ben-marshall, 2020-12-14:
Hi @rdolbeau 

Sorry it's taken me a while to reply.

> So it's likely they end up being implemented in really small RV32 cores... where decoding irregularity is the costliest, I would guess.

I agree. However, I'd balance this against the implementation cost of the entire instruction.

The sort of complexities/costs we're debating introducing into the decoder are dwarfed by the size of the execute part of the instruction. e.g. dozens of extra logic gates for decode v.s. ~1K for the actual instruction implementation.

With this in mind, I'm hesitant to move this complexity into the compiler. In any case, encoding `bs` inside the register numbers puts severe constraints on how these instructions are used in practice. As a more HW orientated person, I'm usually all for pushing complexity into software land, but in this case, I think it is more trouble than it is worth. E.g. It would have to work with the RV32E base architecture, which only has 16 registers etc. This would unnecessarily constrain the RV32I implementations etc etc.

I'd rather RISC-V agree on a proper pattern for destructive instructions, which I think is the globally more appropriate solution.

Cheers,
Ben

ben-marshall, 2020-12-18:
Hello all
The [latest release](https://github.com/riscv/riscv-crypto/releases/) (v0.8.1) updates the AES32/SM4 encodings based on this discussion, and discussions at the regular Task Group meetings.
It still falls foul of some of the points raised here (regularity of field locations, for example) but on the whole the TG think (and I hope you agree) this is a better scheme over all. Or, at least a lesser evil. Please feel free to let me know what you think.
@JamesKenneyImperas  - I'll leave it to you to close this issue if that's okay?
Thanks,
Ben

JamesKenneyImperas, 2020-12-18:
Yes, this looks better.

Thanks!

closed 2020-12-18 by JamesKenneyImperas

rdolbeau, 2021-02-05:
FYI:

a) the new encoding in 0.8.1 for Zkne requires a patch to VexRiscv but not an outlandish one (https://github.com/rdolbeau/VexRiscv/commit/0fd07b48709f45ce0cc88c0c4d54ebb8bc4ed2e0)

b) generator/test codes updated to match 0.8.1 in https://github.com/rdolbeau/VexRiscvBPluginGenerator/, if someone want to try Zkne/Zknh + (most of B 0.94-pre) on RV32