riscv/riscv-crypto #70: Zksd (SM4 decode) has no instructions

rdolbeau, 2021-02-18:
Zksd (SM4 decode) appears in table 1 page 4 as a separate sub-extension & part of Zks, but in table 2 page 5 it has no instructions - the two SM4 instructions (`sm4ed` & `sm4ks`) are in Zkse, and from the look of <https://github.com/rvkrypto/rvkrypto-fips> it seems there is nothing specific to decryption. It seems to me Zksd is not needed?

ben-marshall, 2021-02-18:
Hi @rdolbeau 
You're absolutely right! The two SM4 instructions are used for both encrypt and decrypt so there is no need for a dedicated group. Can't believe I've missed that every time I've read that.

I'd suggest merging `Zkse` and `Zksd` into a single `Zksb` (b for block cipher) to mirror `Zksh` (h for hash function.).

I'll bring it up in the meeting today and leave this issue open until I've committed a fix.

Cheers,
Ben

ben-marshall, 2021-02-22:
We discussed this at the meeting (Feb 18/21) and the fix is to merge into a single group called `Zksed`. I've committed this in d0ae6022 but won't close this issue yet. I'm waiting for any small changes we make up to public review to go in before doing another release. I hope that's alright.
Thanks again for spotting this!

ben-marshall, 2021-03-06:
Closed with `v0.9.0` release.

closed 2021-03-06 by ben-marshall