riscv/riscv-crypto #74: overlapping patterns for `rv32i` and `rv64k`

Lucas-Wye, 2021-02-27:
Hi,

  I am currently working on `K extension`  for `qemu`.

  I met a problem about the instructions `slli` , `aes64ksli` and `aes64im`.
```
FAILED: libqemu-riscv32-softmmu.fa.p/decode-insn32.c.inc
/usr/bin/python3 /home/XXX/qemu/scripts/decodetree.py ../target/riscv/insn32.decode --static-decode=decode_insn32 -o libqemu-riscv32-softmmu.fa.p/decode-insn32.c.inc
../target/riscv/insn32.decode:128: error: overlapping patterns:
../target/riscv/insn32.decode:128: slli 00...... ........ .001.... .0010011
../target/riscv/insn32.decode:604: aes64ks1i 00110001 ........ .001.... .0010011
../target/riscv/insn32.decode:605: aes64im 00110000 0000.... .001.... .0010011
ninja: build stopped: subcommand failed.
Makefile:172: recipe for target 'run-ninja' failed
make: *** [run-ninja] Error 1
```
It seems there are overlapping patterns for `rv32i` and `rv64k`. Besides, `slli` is the same pattern for `rv64i`. So, there are also overlapping patterns for `rv64i` and `rv64k`.

Is there anything wrong with `rv64k`?

grnewell, 2021-02-27:
> EXTERNAL EMAIL: Do not click links or open attachments unless you know the content is safe

Hi ZW,

Thanks for noting this.  I don’t have the answer for you, but we will look into it shortly.

Rich

G. Richard Newell
Assoc. Technical Fellow, FPGA Business Unit, Microchip Technology
(408) 643-6146 (office), (408) 882-4785 (mobile), +1 (925) 478-7258 (Skype)
PGP: (2009 DSA-1024, ELG-4096) B751 FC13 8B4E 49DA 2270 35A2 20E4 E66A D0D0 2E34
     (2016 SSA-4096, RSA-4096) 65F5 CCD6 23B3 BD3D CEDE AB58 171F F4DE E7D0 3ECA

rdolbeau, 2021-02-27:
@Lucas-Wye Seems the pattern for `slli` is not right, from the RISC-V specifications I see from the two relevant chapters:

`0000000 shamt[4:0] src SLLI dest OP-IMM` for RV32I

`000000 shamt[5] shamt[4:0] src SLLI dest OP-IMM` for RV64I

whereas it seems you only have a pair of explicit 0 in your pattern in bit 31 and 30 (I assume the dot `.` is a wildcard).

Same for `srli` and `srai`, they also have 7/6 bits of the immediate fixed, as a ''a specialization of the I-type format.".

ben-marshall, 2021-02-27:
Hi @Lucas-Wye 
I think @rdolbeau  is right, does that fix the problem? Also, there are some instructions in the K extension which deliberately overlap opcodes (e.g. `aes32esi` and `aes64es`), but which only exist in either RV32 or RV64, so do not *actually* cause a problem.
Cheers,
Ben

Lucas-Wye, 2021-02-27:
@rdolbeau @ben-marshall 
Thank you for your answer. I have checked the code. This problem comes from `qemu` and there is something wrong with the instruction decode of `qemu`. Anyway, thanks for all of you.

closed 2021-02-28 by Lucas-Wye