riscv/riscv-crypto #75: "rt" operand didn't actually save encoding points

jnk0le, 2021-02-28:
Destructive "rt" operand was "invented" to reduce opcode pollution, yet RV32 AES implementation still wastes 16 funct7 encoding points (+8 for sm4) at the cost of extra decoder complexity and power draw even for non-K use cases.

And we are talking about risc-v that is known from design decisions around reducing instruction muxing and creating anomalies like jalr bogus jump from lsb+lsb that wasn't even reserved in the ABI. But now we are getting "rt" operand that is gonna be RVK exclusive. Even code size group didn't use such encodings. What could be the `rd!=0` encodings used for if not the RVK 16+8 instructions itself?

Rd==rs1 decision also requires 4 extra move instructions each round, which is [already happening](https://github.com/rdolbeau/VexRiscvBPluginGenerator/blob/master/aes256encrypt-rv32/core.c#L49), or double round unroll happening here in benchmark folder (extra 16+4 instructions are gonna knock out a few cache lines, STM32F4 has 64x16B lines as a reference).

rdolbeau, 2021-02-28:
@jnk0le  I'm not sure what you suggest. Using rd==rs1 (which was also suggested, there's discussion of possible solutions in #65) would presumably cause similar issues, and is worse when it comes to reusing unused encoding space. Basically, you need to somehow specify the two inputs, the output, and the byte select, plus whether you want to do MC. So with no overlap, you need 32[rs1]*32[rs2]*32[rd]*4[bs]*2[mc] encoding point. With the overlap as proposed in 0.8.1, you only need 32[rs1/rd]*32[rs2]*4[bs]*2[mc]. Do you have a proposal to reduce the needed encoding space further?

Also, the 'rt' destructive operand is not completely new, it's already used in CR and CI instruction formats (see table 16.1 page 100 of the unprivileged specs), and in a 4-bits variant in CA as well. But I don't think it would be a good idea to propose that K requires C, and use C instructions for AES/SM4 just so K can reuse an existing destructive format.

ben-marshall, 2021-02-28:
Hi @jnk0le 

You raise some valid points and I'll try to reply to each of them, with apologies to @rdolbeau for repeating some of his points. I'm sorry this reply became much longer than I intended it too. I'm sort of using it as practice for answering questions which will come up at the public review. I'm glad you raised this now!

We certainly discussed these points before settling on the current design. Certainly, the current design is not perfect and involves trade-offs. I'll prefix everything by saying that I think they current design is the best / least worst we could come up with, and obviously if it can be improved, then it should be.

> Destructive "rt" operand was "invented" to reduce opcode pollution, yet RV32 AES implementation still wastes 16 funct7 encoding points (+8 for sm4)

I don't think this is entirely fair, especially with the title of your issue. The current design manifestly saves opcode space, compared to having distinct `rd,rs1,rs2` and `bs` fields. Though, like you say, this is not without it's trade-offs. Like @rdolbeau  says, we are not the first to require rd=rs1, as with the C extension. We *are* the first to require it in a 32-bit instruction, sure. The 16/8 func7 points for AES/SM4 are not *wasted*, they are put to essential use.

> at the cost of extra decoder complexity and power draw even for non-K use cases.

Yes, this is an inescapable side effect of adding instructions. Our current design adds more complexity than we would like, but we can balance this against the total complexity for the entire instruction. As noted in #65, the power/gate costs for decode are small compared to the implementation of the execution logic for the instruction. Yes, the decode logic has an effect on *every* instruction.

> But now we are getting "rt" operand that is gonna be RVK exclusive.

There is no suggestion or requirement that this `rt` operand be RVK exclusive. RVK might be the only extension which uses it *for now*, but having a dedicated format for instructions which can afford to destroy one of their operands is likely useful to other future instructions. Ideally, we would have borrowed an existing instruction format to do this, but since none existed, we made our own which overlapped with the C extension notion of destructive operands as much as possible.

If a new format can be invented which suits our instructions and is more efficient / useful to others (by whatever metric) then we would like to adopt it.

> What could be the rd!=0 encodings used for if not the RVK 16+8 instructions itself?

I won't enumerate hypothetical instructions, but I think it's reasonable to think there is utility in a proper format for instructions which destroy an operand. I don't say our design is the *ideal* format, just that it's the best one we have come up with so far.

> Rd==rs1 decision also requires 4 extra move instructions each round, which is already happening, or double round unroll happening here in benchmark folder (extra 16+4 instructions are gonna knock out a few cache lines, STM32F4 has 64x16B lines as a reference).

Yes. It would be great to avoid this and have the full `aes* rd, rs1, rs2 bs`, but again, we decided that was too big an ask in terms of encoding space. If later on we are told this isn't as much of a concern, then sure, we can go back to this format. That would be great!

I'm actually glad it's possible to change the implementation slightly to better fit within cache-line restrictions. Again, it's not perfect, or as efficient as one would hope, but it is still *much* faster than any software alternative. Evicting a few cache-lines because of these instructions is still a *much* lesser problem than the alternative pure software implementation of AES, which will always be much larger.

So, overall, I accept all of your points. They are very valid and we considered them all during the design of these instructions. A reasonable engineer might have made a different call, but this is the design we felt best balanced all of the trade-offs at play.

I would be very keen to hear if you have a better design in mind that you would be happy to contribute?

Thanks,
Ben

rdolbeau, 2021-02-28:
@ben-marshall Thinking about it, at least for RV32 the reclaimed opcode space from aes32*/sm4* could be reclaimed immediately by fitting every other RV32K instructions in there for RD!=0. There's 24 opportunities (6 instructions, 4 byte select), and only 14 instructions to fit (and some are unary or zero-ary (?) for Zkr). All those instructions are essentially `nop`  when rd==0 anyway.

i.e. a quick'n'dirty fitting could be this:

```
// Current proposal (0.8.1)
11  11000  10101 00000 000   rd  1110011 pollentropy
01  11101  01001 00000 000   rd  1110011   getnoise
00  01000  01000  rs1  001   rd  0010011     sm3p0
00  01000  01001  rs1  001   rd  0010011     sm3p1
00  01000  00000  rs1  001   rd  0010011  sha256sum0
00  01000  00001  rs1  001   rd  0010011  sha256sum1
00  01000  00010  rs1  001   rd  0010011  sha256sig0
00  01000  00011  rs1  001   rd  0010011  sha256sig1
01  01000   rs2   rs1  000   rd  0110011 sha512sum0r
01  01001   rs2   rs1  000   rd  0110011 sha512sum1r
01  01010   rs2   rs1  000   rd  0110011 sha512sig0l
01  01110   rs2   rs1  000   rd  0110011 sha512sig0h
01  01011   rs2   rs1  000   rd  0110011 sha512sig1l
01  01111   rs2   rs1  000   rd  0110011 sha512sig1h
bs  11000   rs2   rt   000 00000 0110011     sm4ed
bs  11010   rs2   rt   000 00000 0110011     sm4ks
bs  11011   rs2   rt   000 00000 0110011  aes32esmi
bs  11001   rs2   rt   000 00000 0110011   aes32esi
bs  11111   rs2   rt   000 00000 0110011  aes32dsmi
bs  11101   rs2   rt   000 00000 0110011   aes32dsi

// Packing everything in aes32*/sm4, somewhat arbitrarily
// Zkr in aes32dsi
00  11101  00000 00000 000 rd!=0 0110011 pollentropy
01  11101  00000 00000 000 rd!=0 0110011 getnoise
// sm3 in sm4ed
00  11000  00000  rs1  000 rd!=0 0110011  sm3p0
01  11000  00000  rs1  000 rd!=0 0110011  sm3p1
// sha256 in aes32esi
00  11001  00000  rs1  000 rd!=0 0110011 sha256sum0
01  11001  00000  rs1  000 rd!=0 0110011 sha256sum1
10  11001  00000  rs1  000 rd!=0 0110011 sha256sig0
11  11001  00000  rs1  000 rd!=0 0110011 sha256sig1
// sha512 in aes32[ed]smi
00  11011   rs2   rs1  000 rd!=0 0110011 sha512sum0r
01  11011   rs2   rs1  000 rd!=0 0110011 sha512sum1r
00  11111   rs2   rs1  000 rd!=0 0110011 sha512sig0l
01  11111   rs2   rs1  000 rd!=0 0110011 sha512sig0h
10  11111   rs2   rs1  000 rd!=0 0110011 sha512sig1l
11  11111   rs2   rs1  000 rd!=0 0110011 sha512sig1h
// sm4*, aes32* untouched
bs  11000   rs2   rt   000 00000 0110011     sm4ed
bs  11010   rs2   rt   000 00000 0110011     sm4ks
bs  11011   rs2   rt   000 00000 0110011  aes32esmi
bs  11001   rs2   rt   000 00000 0110011   aes32esi
bs  11111   rs2   rt   000 00000 0110011  aes32dsmi
bs  11101   rs2   rt   000 00000 0110011   aes32dsi
```

Didn't look if it made sense when considering RV64, and I didn't consider anything other than 'packing' (so decoding might be harder than with the current scheme).

jnk0le, 2021-02-28:
>Also, the 'rt' destructive operand is not completely new, it's already used in CR and CI instruction formats (see table 16.1 page 100 of the unprivileged specs), and in a 4-bits variant in CA as well.

Yes, it's in the C extension, but it's use is balanced by amount of instructions using it, and the C decoder is not affecting 32bit instruction decoding, that's the problem.

> I'm not sure what you suggest.

> I won't enumerate hypothetical instructions, but I think it's reasonable to think there is utility in a proper format for instructions which destroy an operand. I don't say our design is the ideal [...]

I thought more about using one `funct7` encoding and putting the rest of bs/aes/sm4 encoding into "`functrd`". Fitting the rest of rvk int `rd!=x0` (@rdolbeau) is also valid point.

>As noted in #65, the power/gate costs for decode are small compared to the implementation of the execution logic for the instruction

It's still quite small compared to the rest of RV32IMC.

> If later on we are told this isn't as much of a concern, then sure, we can go back to this format. That would be great!

But only if the current form doesn't get frozen. Otherwise there is no hardware benefit from introducing the fuller form (very similar thing to the mask storage issue in RVV).

ben-marshall, 2021-03-01:
I should have said, the crypto extension is currently undergoing the opcode and consistency review, so hopefully we will get some suggested improvements out of that.

> Thinking about it, at least for RV32 the reclaimed opcode space from aes32*/sm4* could be reclaimed immediately by fitting every other RV32K instructions in there for RD!=0. 

> I thought more about using one funct7 encoding and putting the rest of bs/aes/sm4 encoding into "functrd". Fitting the rest of rvk into rd!=x0 (@rdolbeau) is also valid point.

These are both good ideas. I wouldn't be surprised if the opcode review feedback was along the lines of using fewer `func7` encoding points.

> But only if the current form doesn't get frozen.

Indeed. We're a long way from frozen though. And I believe that in the current RISC-V terminology, there are several months and a public review between a specification being "frozen" and "ratified".

Overall, I struggle to know what peoples priorities are with encodings, and where the point between choice A and choice B really lies. I like the two alternatives suggested here, and if the opcode review came back saying we should use them, I'd be happy with that. I'll try and point the rest of the task group to this thread, and leave it open for further discussion.

Thanks a lot,
Ben

rdolbeau, 2021-03-01:
@jnk0le @ben-marshall Is there a RISC-V document describing encoding preferences e.g. between `funct7` and `funct3`?

For instance, it seems K is heavy on `funct7` but light on `funct3`. It could e.g. just as well swap the two MSB for `funct7` with the two MSB from `funct3` to use fewer `funct7`. The only bit change in `funct3` is to identify `sm3*`/`sha256*` from `sha512sum0r`, as they share the same `funct7` (still only looking at RV32), whereas everything else is identified by `funct7`.

From my limited experience as an implementer, as long as all bits are matched separately (i.e. not with rd == rs1 or other 'those two bits must match in the opcode') the encoding was fine - the core design I use handle the complexity for me. Special-casing for `rt` was not an issue.

ben-marshall, 2021-03-01:
@rdolbeau - My impression (which may be false) is that there is lots of rationale for the encodings, some of which is obvious, some of which is not, and some is folklore. E.g. I've never seen a guide which says one should fill up `func3` before filling up `func7`. I think RISC-V generally is still ironing out some of the design guides for new extensions, and scalar crypto (along with Bitmanip) is acting as a pipe cleaner.

rdolbeau, 2021-03-01:
> (i.e. not with rd == rs1 or other 'those two bits must match in the opcode')

... which is something required for the `rd != 0` scheme, unfortunately; matching `00000` is easy, the negation of it is harder. Could be implemented in 5 patterns (i.e. `----1`/`---1-`/`--1--`/`-1---`/`1----`/) I suppose.

rdolbeau, 2021-03-09:
@ben-marshall From the comment in https://github.com/riscv/riscv-opcodes/pull/63#issuecomment-792161574 , it seems using R-format instructions would be acceptable, and that's what I would favor. rd-from-rs1 is working for me as well, but it's a bit of a nuisance to have a special case just for AES32/SM4. Just an opinion from a third-party :-)

ben-marshall, 2021-03-09:
@rdolbeau - I agree, if we can keep things uniform and the wider RISC-V folks are okay with using the R-format, that's what we should do. Here's hoping!

jnk0le, 2021-06-04:
0.9.2 already closes this

ben-marshall, 2021-06-04:
Hi @jnk0le - thanks for your feedback throughout this issue. As you saw, we've gotten rid of the `rt` field for v0.9.2. However, I'd like to keep this issue open until v0.9.2 actually gets released in a week or so. Just so it's more visible for anyone who comes looking for background to the changes in the mean time.
Cheers,
Ben

ben-marshall, 2021-06-11:
Issue closed with release [v0.9.2](https://github.com/riscv/riscv-crypto/releases/tag/v0.9.2-scalar). Please open new issues for subsequent encodings related discussions. Thank you for your input! :+1:

closed 2021-06-11 by ben-marshall