riscv/riscv-crypto #76: Where can we get more information of K ext

VincenttttWu, 2021-03-03:
Hi ben,

It's me again. we have initially implemented the support of K ext instructions on MC layer of LLVM. So we are looking for more information of K ext (such as K ext intrinsic doc). Could you help me?

Sincerely yours,
Vincent

ben-marshall, 2021-03-03:
Hi @VincenttttWu 

Sure thing, glad you asked and great to hear you are making progress!

In terms of intrinsics, the current proposal from @mjosaarinen lives [here](https://github.com/rvkrypto/rvkrypto-fips/blob/main/rvkintrin.h). They are designed to use the same conventions as the Bitmanip intrinsics. It's outside the main riscv-crypto repo so we don't run into license problems getting it upstreamed to GNU GCC. I gather this isn't such a problem for LLVM?

What other information are you after?

Cheers,
Ben

VincenttttWu, 2021-03-03:
Hi @ben-marshall 

That's it. thanks a lot. I've been looking for it for a whole day. 
I think the license may not be a problem for LLVM.
There is no further information required at this time. Thanks again.

Sincerely yours,
Vincent

closed 2021-03-03 by VincenttttWu

grnewell, 2021-03-03:
> EXTERNAL EMAIL: Do not click links or open attachments unless you know the content is safe

Hi Vincent,

Sorry you wasted time looking for the intrinsics proposal.  I have added the link to our on-line status page in the Wiki:
https://wiki.riscv.org/display/TECH/Scalar+Crypto+Standardization+Status+Summary

Hopefully, that will make it easier to find for those that follow in your footsteps.

Rich

G. Richard Newell
Assoc. Technical Fellow, FPGA Business Unit, Microchip Technology
(408) 643-6146 (office), (408) 882-4785 (mobile), +1 (925) 478-7258 (Skype)
PGP: (2009 DSA-1024, ELG-4096) B751 FC13 8B4E 49DA 2270 35A2 20E4 E66A D0D0 2E34
     (2016 SSA-4096, RSA-4096) 65F5 CCD6 23B3 BD3D CEDE AB58 171F F4DE E7D0 3ECA