riscv/riscv-crypto #78: constant time hints are not broad enough

jnk0le, 2021-03-06:
It was decided that constant time execution of multiplies will be governed by rs1/rs2 ordering hints which immediately raises concerns about squaring that requires extra move instructions to allow early-out or mulhsu that can't simply swap arguments and literally requires extra move instruction for both; constant time and early outs.

That may be a little inconvenience, but what with the other instructions that also can't have arguments swapped and are subject to data dependent optimizations like `cmov` and `cmix` (aka `BPICK` in P ext) instructions - those when implemented on simple 2R1W regfile based cpu can read the RS3 in second cycle but they don't even need to, if rs2 evaluates to pick only rs1 in first cycle. Furthermore those instructions have a lot of non-crypto use cases where forcing worst case scenario for every `cmov`/`cmix` is gonna diminish performance gains.

There are also various P ext instructions that might need to be covered. Otherwise we will end up with cortex m3 long multiply situation (umull etc.). Not everything can have swapped arguments.

In RVV there is way more trickery to defuse. I made a post about it here riscv/riscv-v-spec/issues/417
Original proposal uses bit in vtype but it can be a separate csr as well, especially when it will use more than just one "constant time" bit to fine grain the constant'd operations with FENCE like fall back. Proposal for upper `vtype` retainer bit in `vtypei`  was rejected due to implementation complexity, so the separate CSR might actually be better approach in this case.

I assume that floating point was intentionally, explicitly left out of spec as it doesn't have use in crypto, right?

ben-marshall, 2021-03-06:
As written, our mechanism for indicating constant time behaviour applies _only_ applies to instructions talked about in the scalar crypto extension. It does not affect `cmov`, `cmix` or anything else. We don't require it for anything in the P extension either. All of these are out of scope right now.

If you've been given the impression our mechanism does apply to these instructions, please let me know where/how so I can make it more clear.

We deliberately excluded `cmov` and `cmix` from our spec a long time ago to avoid requiring three source registers in any instructions, accepting that there might be a performance  / efficiency cost to that. To do `cmov` style functions, we'd recommend building a mux using the `andn` instruction.

About using a CSR and a single "constant time mode bit". I think this would actually be a nicer solution to use across the ISA, so long as you could be certain exactly which subset of instructions the CSR controlled. Like I said in the other issue, we rejected that because we expected adding state (or instructions) to cause more protest than the current mechanism, but if we judged that wrong, then it can be changed.

I agree that the V extension is a nightmare in terms of constant time behaviour, but that again is out of scope for now in trying to get the scalar spec finished. It will be back in scope for the vector crypto extension. It would be great if we can use a global mechanism for indicating constant time mode (like a CSR) but that will require input from a lot of people. Coaxing people to attend our meetings is hard, so it might come out at public review.

Either way, I'm glad you raised this now, especially in the vector thread. People's replies there have been very illuminating.

Cheers,
BenRight now, we have a small, core set of essential instructions which nearly everyone implements (the M extension) which we are trying to guarantee behaviour for under cryptographic workloads. The P extension might be useful for cryptographers, but we aren't in a position right now to start putting constraints on its behaviour for all sorts of reasons. Not least trying to get _something_ finished. If you think our specification needs to be clearer about exactly what it is promising, then let me know and I'll update the language.

jnk0le, 2021-03-08:
>If you've been given the impression our mechanism does apply to these instructions, please let me know where/how so I can make it more clear.

>We deliberately excluded cmov and cmix from our spec a long time ago to avoid requiring three source registers in any instructions, accepting that there might be a performance / efficiency cost to that. To do cmov style functions, we'd recommend building a mux using the andn instruction.

I'm not advocating for mandatory inclusion of `cmov`/`cmix` in K extensions. It's more about situation where those instructions are available due to their cheapness of implementing on specific cores or just the mandatory availability (P long mult/mac).

Even the multi-cycle worst case scenario for those instruction is still faster than implementation based on "crypto-recommended" instructions. In case of high perf cores with single cycle ternary and IFMA we quickly get into "why should I use inferior instruction sequence when the better one is still constant-time" problem.

ben-marshall, 2021-03-09:
I've just sent [this message](https://lists.riscv.org/g/tech-crypto-ext/message/451) to the public crypto task group mailing list with the aim of getting wider input on all of this.

ben-marshall, 2021-06-11:
I'm going to consider this issue closed with release [v0.9.2](https://github.com/riscv/riscv-crypto/releases/tag/v0.9.2-scalar). Please open new issues for subsequent constant-time related discussions. Thank you for your input! :+1:

closed 2021-06-11 by ben-marshall