riscv/riscv-crypto #80: Merge error in the file `https://github.com/riscv/riscv-crypto/blob/master/tools/patch-binutils.patch`

Lucas-Wye, 2021-03-09:
Errors in the file `https://github.com/riscv/riscv-crypto/blob/master/tools/patch-binutils.patch`

ben-marshall, 2021-03-09:
Hi @Lucas-Wye 
What errors are you seeing? Can you paste the output of "git apply" when applying the patch?
Thanks,
Ben

ben-marshall, 2021-03-09:
What is the output of running `git log -n 5` inside `riscv-crypto/extern/riscv-gnu-toolchain/riscv-binutils`?
I'm trying to work out if the `start-from-scratch` script is checking out the right base commit for binutils. It *should* be `3e5f50f31348d05144966545d862a3172d315230`, but I see now it's actually checking out a branch pointer, which might have changed recently.

Lucas-Wye, 2021-03-09:
[image]
Yes, the commit id is wrong.

ben-marshall, 2021-03-09:
Ah, that explains it then. I've just run a clean checkout and start-from-scratch and it is working for me. Let me know if changing the commit makes it work for you?

Lucas-Wye, 2021-03-09:
Thank you! It works for me now.

closed 2021-03-09 by Lucas-Wye

ben-marshall, 2021-03-09:
Glad to hear it! :-)