riscv/riscv-crypto #82: Issues with rcon

jrtc27, 2021-03-11:
1. `rcon` should be `round`/`round_number` (or an abbreviated form thereof), since it's the round number used to derive the round constant (the two are equivalent but not the same)
2. The spec permits 0<=`rcon`<=0xA, but `rcon[i]` is only defined for `i >= 1`, so the spec should clarify exactly how the (renamed) `rcon` relates to the AES specification

ben-marshall, 2021-03-11:
Hi @jrtc27 

> rcon should be round/round_number (or an abbreviated form thereof), since it's the round number used to derive the round constant (the two are equivalent but not the same)

Thanks for this. I agree, if it refers to the round *number*, it should be called `rnum` or some-such.

> The spec permits 0<=rcon<=0xA, but rcon[i] is only defined for i >= 1, so the spec should clarify exactly how the (renamed) rcon relates to the AES specification

Also agreed. I can add a simple table/mapping to the spec and a note to explain how it relates to the FIPS 197 AES spec, and can point to the Sail code in the appendix.

I'll raise both of these at the meeting. Any changes will show up in a commit, but might not make it into a proper release for a little while. I'll be sure to leave this issue open until a release is made with the changes.

Cheers,
Ben

ben-marshall, 2021-03-17:
Okay, we discussed this in the meeting last Thursday and agreed it merits tidying up.

I think the simplest solution to this will be to rename `rcon` -> `rnum`. A sentence in the spec saying that "the round number provided by the immediate is translated into the corresponding round constant using the following mapping..." should then suffice. I'll add comments in where relevant about which part of the FIPS 197 spec it relates to. I think the original reason the immediate was called `rcon` was because that's what the "round constant word array" is called in FIPS 197.

The Sail function `aes_decode_rcon` can be renamed `aes_rnum_to_rconst` (or even be turned into a mapping if that makes more sense).

There are a few pending changes building up for the next release, so I'll roll this change together with those, since the Sail/Spike/Spec/Opcodes repos will need changing and it's easier for me to do everything at once.

Thanks again,
Ben

jrtc27, 2021-03-17:
Great, all sounds sensible, thanks.