riscv/riscv-crypto #83: Improve friendliness of benchmarks?

JamesKenneyImperas, 2021-03-11:
I've been trying to use the benchmarks included here. I think it would be helpful if they could be cleaned up somewhat so that it is more obvious to less-experienced users what they are trying to show.

For anything demonstrating an encryption/decryption flow, what I really want to see as a user is:
1. The plain text (before encryption);
2. The encrypted text.
3. The plain text (after decryption) which should of course match 1.
Any other state that is dependent on the precise implementation of the algorithm (e.g. instruction counts) should not be shown, unless clearly labelled as implementation-dependent. It is also desirable that the same data is used by each benchmark whichever algorithm is used.

When I look at the output from the aes_128_* tests, it doesn't seem that they are in a helpful form like this.

Firstly, it is not clear what is being reported: various lines are reported with titles key, erk, drk, pt, pt2 and ct, but it isn't obvious to the uninitiated what these are.

Secondly, it appears that the output differs depending on the selected implementation: the "reference" implementation does not produce the same results as the "zscrypto" implementation. After some investigation, it seems that this is because the _input_ on each test loop is determined by a random number sequence based on current time. I think it would be much more useful to feed the same pseudo-random sequence through each iteration, irrespective of the implementation. This makes it much clearer to a user that the implementations are doing the same thing.

Thirdly, it seems that the reported drk values depend on the implementation. It is confusing to print this data without making it clear that this is implementation-dependent internal state.

I haven't looked in detail at other benchmarks here, but scanning the output make me think they have similar issues.

Thanks.

ben-marshall, 2021-03-11:
Hi @JamesKenneyImperas 

I think you've captured something which has been looming for a while. I agree that the benchmarks could do with 
- A general tidyup of the source code and makefile infrastructure
- Some additional printing of input/output etc as you describe
- Certainly having the reference and zscrypto algorithms use the same inputs / outputs is a really sensible change.

In terms of fixing this:
- @mjosaarinen has made an awesome effort of expanding and increasing the quality of benchmarks using the scalar crypto extension in a separate repository. These can be found [here](https://github.com/rvkrypto/rvkrypto-fips). I think they well address all of your concerns. You might prefer to just look at them going forward? I should add a better signpost to them from the README.
- The state of the benchmarks in the riscv-crypto repo is my fault to fix. I would be *tempted* (particularly in time for public review) to actually replace the benchmarks in the riscv-crypto repo with a submodule pointing to Markku's and to include a flow for building and running them on Spike.

Having separate / independent sets of benchmarks is nice, but the ones in riscv-crypto have largely been superseded by Markku's, and I'm unlikely to be able to write an independent set of the same quality in any time frame.

I'll bring it up for discussion at the meeting later. I think having a set of "friendly" benchmarks is certainly really important for public review.

I hope all of that makes some sense?

Cheers,
Ben

JamesKenneyImperas, 2021-03-11:
Hi Ben,

Thanks for the response. We'll take a look at Markku's work, and also see what comes out of your meetings.

James.

rdolbeau, 2021-03-13:
@JamesKenneyImperas @ben-marshall I think that currently @mjosaarinen code is for validation but does not yet include performance measurement.

The benchmarks from this repo measure instructions count by default, but can easily be fixed to use cycles count instead, the infrastructure is there in the headers, you can just do e.g.:
`sed -i -e 's/test_rdinstret()/test_rdcycle()/g' benchmarks/test/*`
and recompile. For either to work the relevant performance instruction need to be supported by the core.

Example (first cycle, then instruction):
```
bench_cyc/test_block_aes_128-aes_128_zscrypto_rv32.elf
aes_128_zscrypto_rv32 AES 128 Test passed. enc: 831, dec: 571, kse: 510, ksd: 1387, 
aes_128_zscrypto_rv32 AES 128 Test passed. enc: 796, dec: 608, kse: 553, ksd: 1423, 
aes_128_zscrypto_rv32 AES 128 Test passed. enc: 822, dec: 611, kse: 455, ksd: 1424, 
(...)
bench_inst/test_block_aes_128-aes_128_zscrypto_rv32.elf
aes_128_zscrypto_rv32 AES 128 Test passed. enc: 241, dec: 243, kse: 221, ksd: 733, 
aes_128_zscrypto_rv32 AES 128 Test passed. enc: 241, dec: 243, kse: 221, ksd: 733, 
aes_128_zscrypto_rv32 AES 128 Test passed. enc: 241, dec: 243, kse: 221, ksd: 733, 
(...)
```

The codes I have for RV32 in https://github.com/rdolbeau/VexRiscvBPluginGenerator/ might also be of interest; they are stand-alone codes designed to work with the [Supercop benchmark](http://bench.cr.yp.to/supercop.html). They copy the needed infrastructure for the 'try' binaries, but the implementation files should be usable from inside the supercop benchmark (I haven't tried that yet, as my FPGA infrastructure is a bit small to handle the full Supercop).

kdockser, 2022-10-29:
This is left over from Scalar Crypto which has already been ratified. While someone can improve the friendliness of the scalar benchmarks, this is no longer an issue that needs to be tracked here.

closed 2022-10-29 by kdockser