riscv/riscv-crypto #88: Register index ordering vs. CSR bitmap for constant-time latency constraints.

WildCryptoFox, 2021-04-16:
#64 mentions a single bit CSR was considered but decided against due to the introduction of state; without referencing any such discussion. Where was this discussion?

I find this result surprising because any compiler not aware of the K extension will produce unnecessarily constrained code. A K-aware compiler doesn't win either, as it will not always be able to avoid adding unnecessary constraints and will need additional copies to enforce constraints.

I expected a CSR bitmap over the X (and V) registers. Yes this is additional state, but it is only 32 (or 64) bits -- or fewer if the implementation opts for a coarser granularity. 16, 8, 4, 2, or 1.

RVE implementations, where size really matters, could use the unused high register bit to signal the constant-time constraint without paying for an additional CSR.

Unfortunately the current RV32E draft explicitly marks the unused high bit as "custom" and attests that no future standard will ever use it.

>  Any future standard extensions will not make use of the instruction bits freed up by the reduced register-specifier fields and so these are designated for custom extensions.

Although a size-optimized core may not pay the extra area for variable time optimizations. Energy-optimized cores might, thus RV32E remains relevant.

ben-marshall, 2021-04-16:
Hi there

For the sake of others reading this, I'll repeat some of my response to #87 

> We are in the process of revising our constant time proposals, the latest draft of which is available [here](https://github.com/rvkrypto/riscv-zkt-list/blob/main/zkt-list.adoc). It includes a much larger list of core instructions, and aims to be much simpler. That is, you either are constant time or not. You can't switch it on or off.

About your points here:

> Where was this discussion?

Mostly in our weekly online meetings. If you are on the tech-crypto mailing list for RISC-V, you'll be able to search there for other discussions and for our minutes. The list is public to view but you must be a RISC-V member to post I believe.

> I expected a CSR bitmap over the X (and V) registers. Yes this is additional state, but it is only 32 (or 64) bits -- or fewer if the implementation opts for a coarser granularity. 16, 8, 4, 2, or 1.

Do you mean an entire new CSR to indicate which GPRs or vector registers, when sourced, must execute in constant time? This has an _enormous_ number of implications for more complex micro-architectures which I'm sure would be very unpopular. You would have to tag every physical register with extra bits related to whether they need constant time properties or not.

> RVE implementations, where size really matters, could use the unused high register bit to signal the constant-time constraint without paying for an additional CSR.
> Although a size-optimized core may not pay the extra area for variable time optimizations. Energy-optimized cores might, thus RV32E remains relevant.

RVE is an interesting one. The reduced number of registers hugely increases the static and dynamic code size, meaning you need larger memories to store the code, which take more space and can negate the area savings of RVE in the core. Having to fetch more instructions totally removes any energy savings you might otherwise have had. I can't speak to the general relevance of RVE without more specific examples. For the cryptographic workloads we are concerned with, it would be a performance and code size disaster due to the large state sizes.

I'm not sure if I can quite parse a specific question or suggestion in your comment otherwise? But I'm glad you are asking nonetheless, feedback is always helpful.

Cheers,
Ben

WildCryptoFox, 2021-04-16:
> Always on. [..] where -> mailing lists

Hmm. Always constant does solve the issues.. However painful forbidding vartime optimizations for non-sensitive code.

> CSR

Yes, a new CSR tagging each bit (or coarser, implementation decided partitions). Could you elaborate the costs? If the costs are too great then I guess the best we can have is coexisting cores with and without the Zkt extension. Which would require operating system enforced policies instead.

> RVE

That makes sense.. Probably no RVEZkt then.

> I'm not sure if I can quite parse a specific question or suggestion in your comment otherwise? But I'm glad you are asking nonetheless, feedback is always helpful.

You answered my main question (where the discussion was). Just a matter of searching through these archives. If anyone can provide links to specific emails, I'd appreciate it.

I'm working towards a compiler and language which tracks data-dependent classifications to constrain the optimizer -- without effectively disabling the optimizer while using user-defined domain-specific optimizations. The always-constant constraint suffices for verification. Along with [seL4's `fence.t` instruction](https://arxiv.org/abs/2005.02193).

Any suggestions for my project would be appreciated, though any thorough discussion would be off-topic. Feel free to join `#wild` on Freenode.

Thanks for the quick responses.

ben-marshall, 2021-04-16:
> Yes, a new CSR tagging each bit (or coarser, implementation decided partitions). Could you elaborate the costs? If the costs are too great then I guess the best we can have is coexisting cores with and without the Zkt extension. Which would require operating system enforced policies instead.

So, I''m not such an expert on high performance architectures, but anything which requires tracking of bits associated with registers files becomes expensive, because super-scalar out of order machines have very large physical register files, with many physical registers per architectural register. So, adding one 32-bit register, with one bit per _architectural_ register, might require adding an extra bit to >100 _physical_ registers to track the same information over time, with associated area/performance costs.

This is only my intuition though, with more work, it might be possible to remove this overhead and I'd be happy to be told by someone more knowledgable that is the case.

Remember though, Zkt is really a simplicity driven proposal, and does not preclude other future mechanisms (which would likely be mutually exclusive to Zkt) for turning constant time guarantees on or off. We're simply prioritising the thing people need/want right now.

> Any suggestions for my project would be appreciated,

There is work to standardise something to the effect of `fence.t` within the RISC-V ISA. If you are able to become a member (it's free) you can join in. Navigating RISC-V is hard, but the website provides a reasonable breakdown of who is working on what. Plus there's the [wiki](https://wiki.riscv.org).

Cheers,
Ben

WildCryptoFox, 2021-04-16:
> might require adding an extra bit to >100 physical registers to track the same information over time

Oh. I wouldn't use hardware level taint analysis, so the physical registers don't need a tag. I should have elaborated, sorry.

Let there exist a new CSR with 32 bits (for RVI, another 32 for RVV). The Nth register is sensitive if the Nth bit of the mask is high. Because the zero register is constant (therefore public), we're safe to repurpose its bit to decide if we want to trap on sensitive loads. Assuming a mask set to `0xFFFF0000`, we do not trap on illegal loads, registers `x1-x15` are public, `x16-x31` are secret. `ADD` decodes to `CT-ADD` IFF `mask[rs1] or mask[rs2]`. Similarly `LB` decodes to `CT-LB` IFF `mask[rd]`. `LB` traps if `mask[0] and mask[rs1]`.

`ADD` does not set `mask[rd] = mask [rs1] or mask[rs2]`, as that'd implement taint tracking. OTOH [CHERI-RISC-V](https://www.cl.cam.ac.uk/research/security/ctsrd/cheri/cheri-risc-v.html) exists and does pay for tags on physical registers.

A trivial implementation acts as Zkt would under the new draft, plus a user-mode read-write CSR hardcoded to 0xFFFFFFFF. The program may read the CSR back to see if the core allows mixing constant-variable code, possibly deciding to use different code.

RV32KV would need a second CSR for the vector classification mask. RV64KV could use 64-bits in a single CSR.

> does not preclude other future mechanisms (which would likely be mutually exclusive to Zkt)

If there were a fine-grained alternative I don't think it'd need to be mutually exclusive. It'd just need to act like Zkt in its default state.

> There is work to standardise something to the effect of fence.t within the RISC-V ISA.

I've heard from a seL4 talk there is an effort but I haven't seen it. Searching that wiki yields no results. Do you have a link?

> If you are able to become a member (it's free) you can join in.

[ ] TODO: Become a RISC-V member. :smile:

ben-marshall, 2021-04-20:
This scheme is _very_ complex. And considering how un-popular my strawman proposal of having only a single CSR bit to turn constant-time on/off, I think the rest of the RISC-V world would take a lot of convincing. And that's not to say whether you need the extra CSRs in your proposal to be saved on a context switch / during an exception / interrupt.

I might be wrong, but I think your scheme still needs a tag bit for physical registers, or at the very least, an extra tag bit for every instruction in the re-order buffer or a reservation station or anywhere else in the execution pipeline. My understanding of these components is that _any_ extra state you add will have an outsized effect.

> If there were a fine-grained alternative I don't think it'd need to be mutually exclusive. It'd just need to act like Zkt in its default state.

Yeah, that makes sense.

> I've heard from a seL4 talk there is an effort but I haven't seen it. Searching that wiki yields no results. Do you have a link?

I think the effort is just getting going, so there is no formal list on lists.riscv.org yet. If you can become a member now and subscribe to the security list, you'll probably hear about it soon.

Cheers,
Ben

WildCryptoFox, 2021-04-20:
At least there is a well established dirty-bit status protocol introduced for floats and copied by vectors to reduce context switching overheads.

If by "extra tag bit" you mean to invalidate the specific decoded instructions which would become invalidated by updating the mask on an out-of-order core; I just wouldn't do that. Updating the mask should be rare, so I'd expect it acceptable to flush the whole pipeline.

In any case; I'm happy with the "always constant-time" approach of the latest draft. I'd rather see K-compliant cores sooner than wait for a weakening protocol to allow data-dependent optimizations for non-sensitive data -- however that may end up implemented. Worst case mix cores where only a subset are K-complaint and leave the policy to the operating system.

I've registered for membership and now just waiting for approval. I might subscribe to some lists.

ben-marshall, 2021-06-11:
I'm going to consider this issue closed with release [v0.9.2](https://github.com/riscv/riscv-crypto/releases/tag/v0.9.2-scalar) which contains the newly expressed `Zkt` extension. Please open new issues for subsequent constant-time related discussions. Thank you for your input! :+1:

closed 2021-06-11 by ben-marshall