riscv/riscv-crypto #89: Specification document - LaTeX to Asciidoc conversion

ben-marshall, 2021-04-21:
The specification needs to be converted from LaTeX to asciidoc, specifically the [asciidoctor](https://asciidoctor.org/) tool. This issue tries to capture what needs to happen.

## What needs converting?

All of the LaTeX files under `doc/spec/tex`. Starting with the `scalar` and files shared between specifications, but moving onto the vector spec.

## How will it be converted:

We can use [Pandoc](https://pandoc.org) to bulk convert the files automatically, but there are several important gottchas while will need manual or semi-automatic fixing:

- We have an extensive bibliography, particularly for the Entropy Source which should be preserved. Asciidoc does not have such a powerful bibliography system as LaTeX, but there _is_ a standard way to do it described [here](https://docs.asciidoctor.org/asciidoc/latest/sections/bibliography/). There is also a WiP script (`bin/bibtex-to-asciidoc.py`) to automatically convert the bibtex citations to the asciidoc format.

- Pandoc _does not_ convert the latex `\cite{X}` macro into an asciidoc reference. The entire citation simply disappears from the output. This will either need to be fixed manually, or semi-scripted to be fixed.

- We currently inline the Sail code describing the functionality of our instructions into the specification, by sourcing from the Sail files directly. Pandoc is smart enough to extract the right parts of the source code in its automatically converted output. However, this will need to be manually maintained whenever the Sail code is updated, unless an automated solution exists / can be found.

- Pandoc mangles anything in the verbatim-esq `cryptoise` environment. This can be manually fixed. It's just for showing assembly syntax.

## Open Questions:

- It's not clear how to represent the encodings? We can generate these tables if need be, as is currently done for the LaTeX, though it would be good if there was a standard way to do this.

- We currently have one monolithic specification document, which works really well as a "single source" for people to look at. It's likely that this will be split into separate components:
  - The `ZKb` section, which is just a list of instructions from the Bitmanip extension which are useful for Scalar Cryptography, along with some rationale.
  - The dedicated _scalar cryptography_ instructions, which exist only in this extension.
  - The entropy source.
  - The `ZKt` proposal, which is orthogonal to the rest of the spec, and adds constant time guarantees to listed instructions. It is _already_ in asciidoc, and is maintained [here](https://github.com/rvkrypto/riscv-zkt-list). It will eventually be moved into this repository.

- It's not entirely clear whether these should be distributed as a single document, or maintained as separate files but combined into one monolithic document, as is currently done with the LaTeX sources. In any case, all of them will need to be versioned.
  - For the purposes of making progress and minimising the amount of change as we go, it might be sensible to _first_ create a monolithic document with asciidoc as the source, _and then_ break it into components for the purpose of ratification.

## Progress

**Shared files:**

File | Automated Conversion | Manual Fixes | Notes
-----|---------------------------|------------------|-----------
appx-benchmarking.tex           |     |     | Doesn't need converting. Not maintained.
appx-materials.tex          | [ ] | [ ] |
changelog.tex              |     |     | Doesn't need converting. Not maintained.
contributors.tex           | [ ] | [ ] |
sec-audience.tex           | [ ] | [ ] |
sec-policies.tex           | [ ] | [ ] |
sec-sail.tex            | [ ] | [ ] |
sec-scalar-profiles.tex    | [ ] | [ ] |

**Scalar Crypto:**

File | Automated Conversion | Manual Fixes | Notes
-----|---------------------------|------------------|-----------
sec-scalar-intro.tex       | [ ] | [ ] |
sec-scalar-aes.tex         | [ ] | [ ] |
sec-scalar-sha2.tex        | [ ] | [ ] |
sec-scalar-sm3.tex         | [ ] | [ ] |
sec-scalar-sm4.tex         | [ ] | [ ] |
sec-scalar-timing.tex      | [ ] | [ ] |
appx-scalar-encodings.tex  | [ ] | [ ] | Need some guidance on standard way to do this.
appx-scalar-sail.tex      | [ ] | [ ] |

**ZKb instruction group:**

File | Automated Conversion | Manual Fixes | Notes
-----|---------------------------|------------------|-----------
sec-scalar-bitmanip.tex    | [ ] | [ ] |

**Entropy Source:**

File | Automated Conversion | Manual Fixes | Notes
-----|---------------------------|------------------|-----------
sec-entropy-source.tex     | [ ] | [ ] |
appx-entropy.tex       | [ ] | [ ] |

**Vector:**

These are not a priority right now, we are focusing on the scalar crypto work.

File | Automated Conversion | Manual Fixes | Notes
-----|---------------------------|------------------|-----------
sec-vector-aes.tex         | [ ] | [ ] |
sec-vector-clmul.tex       | [ ] | [ ] |
sec-vector-grev.tex        | [ ] | [ ] |
sec-vector-intro.tex       | [ ] | [ ] |
sec-vector-profiles.tex    | [ ] | [ ] |
sec-vector-rotate.tex      | [ ] | [ ] |
sec-vector-sha2.tex        | [ ] | [ ] |
sec-vector.tex        | [ ] | [ ] |
appx-vector-encodings.tex | [ ] | [ ] |

elisa-riscv, 2021-04-21:
In answer to your open question about whether to have a monolithic document or to break it into sections, I recommend _chapter-sized_ chunks. The chapter-sized chunks can be grouped into sections, and all can be included in a book header file. You can find an example at https://github.com/riscv/docs-templates. Right now some specifics are subject to change and there is a bug with respect to the use of wavedrom diagramming for which we might need to provide a temporary workaround.

ben-marshall, 2021-04-28:
Hi @elisa-riscv 

I've made a start translating some parts of the specification which are very stable just to try things out. Your template repository and guide have been extremely helpful!

The asciidoc source so far can be seen [here](https://github.com/riscv/riscv-crypto/tree/dev/next-release/doc/adoc-scalar) and I've attached a working copy to this comment ([riscv-crypto-spec-scalar.pdf](https://github.com/riscv/riscv-crypto/files/6392560/riscv-crypto-spec-scalar.pdf)). Without wanting to take up too much of your time, I hope I'm going in the right direction?

Cheers,
Ben

elisa-riscv, 2021-04-28:
HI Ben,

Definitely you are going in the right direction. I am glad that you've got
a build working for you and are on your way.

Just a couple of quick comments, after a quick look:

- the pdf to which you pointed does *not* make use of the fonts and styles
that I provided. I can imagine that you want to test that your conversion
builds in the default pdf first--a good idea--but at some point you'll want
to ensure that the customized look and feel builds properly.
- I'm not certain about your heading use, because in general I am expecting
that topics start with a heading 1, indicated by `==`,  but some document
structure and conventions are still under discussion.

Do let me know if you run into any issues.

I'm thinking that I should put together a checklist for people to use as
they go through the conversion process.

- Elisa

ben-marshall, 2021-06-04:
Closed with release v0.9.1

closed 2021-06-04 by ben-marshall