riscv/riscv-crypto #90: mentropy/pollentropy privilege levels

chjj, 2021-05-23:
From the spec:

> The main ISA-level interface consists of a single pseudoinstruction, `pollentropy` that returns a 32/64-bit value in a CPU register. It is invoked in _Machine Mode_ (which may be the only the mode) as follows: ...

Is there a reason `mentropy` is a machine-level register? This seems to be contrary to every other ISA which has some kind of entropy instruction or register:

- x86's [rdrand]/[rdseed] instructions are available at all privilege levels.
- power's [darn] instruction is unprivileged.
- aarch64's [RNDR]/[RNDRRS] registers are accessible from EL0.
- VIA's [xstore] instruction can be used at any privilege level.
- z/arch's [PRNO] instruction can be used from user-level code (can't seem to find explicit docs on this, but here is IBM's official [zoslib] executing it from user code as proof).

RISC-V appears to be the odd man out.

I understand that `mentropy` is probably meant for seeding a kernel's RNG, but it is still useful to seed/supplement a user RNG with hardware entropy. A few prominent crypto libraries do this ([libsodium] for example).

[rdrand]: https://www.felixcloutier.com/x86/rdrand
[rdseed]: https://www.felixcloutier.com/x86/rdseed
[darn]: https://openpowerfoundation.org/?resource_lib=power-isa-version-3-0
[RNDR]: https://developer.arm.com/documentation/ddi0595/2021-03/AArch64-Registers/RNDR--Random-Number
[RNDRRS]: https://developer.arm.com/documentation/ddi0595/2021-03/AArch64-Registers/RNDRRS--Reseeded-Random-Number
[xstore]: http://linux.via.com.tw/support/beginDownload.action?eleid=181&fid=261
[PRNO]: https://www.ibm.com/support/pages/system/files/2020-07/SA22-7871-10.pdf
[zoslib]: https://github.com/ibmruntimes/zoslib/blob/6493e33/src/zos.cc#L2746
[libsodium]: https://github.com/jedisct1/libsodium/blob/d54f072/src/libsodium/randombytes/internal/randombytes_internal_random.c#L582

ben-marshall, 2021-05-24:
Hi Christopher

> I understand that mentropy is probably meant for seeding a kernel's RNG,

This is the core reason. The RISC-V *entropy source* is just that, an entropy source, as opposed to a *random number generator* with (in some cases) inscrutable guarantees of entropy provided by other ISAs. So yes, the RISC-V proposal is the odd one out, deliberately so.

Our expectation is that if (say, on Linux) you need entropy, you should be going to `/dev/[u]random`, and it is the Kernel's responsibility to seed this with `mentropy` if it is available.

Indeed, it is useful to give user-level applications and libraries access to hardware entropy. _However_ our brief inside the cryptography task group was a cryptographic entropy source only, not a generic source of hardware assisted random numbers. So, we have tried to stay very tightly inside our brief (because getting anything todo with entropy ratified and adopted is hard enough!). 

Also remember that this solution for RISC-V needs to work for cores from tiny, smartcard/micro-controller cores all the way to Big Iron and server class machines. This is not the case for all of the examples from other ISAs that you list, at least to the best of my knowledge. `rdrand` for example uses an entire AES engine, which is unacceptable for us to code into the ISA.

That doesn't mean that in future, there can't be an extension of this mechanism (or a new one entirely) to give user-level code access to hardware entropy. You're not the first to query this and I'm sure you won't be the last. So I think it's fairly likely this will happen at some point if the right people come together to drive it.

I hope this makes sense? Please let me know if not and I'll try again. Questions like this are really useful for us because it helps us practice explaining things prior to public review.

Cheers,
Ben

mjosaarinen, 2021-05-24:
Hi,

In addition to what Ben said, NIST SP 800-90B prohibits "adversarial observation" of the noise source, and to do that it is best to restrict access to the entropy source too ("greatest extent possible"!). Section 3.2.2:

[image]

This is the current standard, published in 2018, and validation to it started last November. The CMVP is no longer accepting submissions to FIPS 140-2 modules that complied with the older random number standards. Some instruction sets mentioned are a bit obsolete now from the viewpoint of usage in securite modules. 

Note that the nice Intel instructions can't be used in FIPS cryptography without help from Intel (putting the Intel chip into debug mode). Also RDRAND is of course often limited to 128-bit security level; not good enough for many applications.

We can probably make a DRBG output  ( 90C terms RBG2(P) and DBG3 ) available in U-mode, but since those require a full AES block, that was left out of the scalar extension. But that will of course be an another instruction. For now we at least have an instruction that can seed the kernel pool quickly and userland can use that. The same minimalistic instruction also works well in security controllers, as Ben noted.

Cheers,
- markku

repnop, 2021-05-30:
Is there a reason to not allow this in S-mode either? It seems a bit awkward to need to have an SBI extension for pulling the value from `pollentropy` only to then possibly need to `wfi`-loop and re-call the SBI function which will go through a lot of churn to simply execute a single instruction on the same hart and return it to S-mode.

mjosaarinen, 2021-06-04:
The CETG agreed that S-mode is okay. There's a PR with that effect in: https://github.com/riscv/riscv-crypto/pull/91

Should be there as we go into public review. I'm closing this issue as it refers to an older version of the specification.

Cheers,
- markku

closed 2021-06-04 by mjosaarinen