riscv/riscv-crypto #92: some links broken in github adoc viewer as well as pdf

jnk0le, 2021-06-11:
affected sections (as in pdf)
2.1
2.2
2.3
5.4.1
5.4.2
5.4.3
5.4.5

e.g.
In github viewer clicking on `[insns-ror]`, appends the `#insns-ror` to the address with no change to displayed content.
In pdf `[insns-ror]` also points to `[...]riscv-crypto-spec-scalar-v0.9.2.pdf#insns-ror`, with no effect after clicking it.

Even if it's intentional to not lead anywhere outside document, I think those shouldn't be made as a pseudo-clickable links.

ben-marshall, 2021-06-14:
Hi there

Thank you for writing this up. Yes, this should have been documented more clearly and has caused no end of confusion (my bad). Eventually, these links will point to instruction descriptions in the big monolithic spec document, but some of these instructions exist in bitmanip, and some don't even have a proper definition in Bitmanip yet (though [this PR](https://github.com/riscv/riscv-bitmanip/pull/134) tries to address that).

So yes, this will be fixed in the next release, where I'll just duplicate the information in bitmanip in the scalar crypto document, so we don't need to keep flicking between specifications, or link across documents.

I'm sorry the communication around this was so bad!

Thanks,
Ben

ben-marshall, 2021-09-08:
All the links that _can_ be fixed for the public review are now fixed, so I'll close this one for now. Thanks again!

closed 2021-09-08 by ben-marshall