riscv/riscv-crypto #97: aes64ks1i rnum argument legal values

JamesKenneyImperas, 2021-06-14:
Hi @ben-marshall ,

I seem to remember from older versions of this specification that the `rnum` operand to `aes64ks1i` was restricted to the legal range `0x0-0xa`, and that values `0xb-0xf` were undecoded and therefore Illegal Instructions. Is it now the case that `0xb-0xf` are treated like `0xa` and not illegal? Or have I misremembered what was previously specified?

Thanks,

James.

ben-marshall, 2021-06-14:
Hi @JamesKenneyImperas 

Nope, you remember correctly, it looks like this is another thing which got lost and missed in the LaTeX -> Asciidoc transition for v0.9.1. Values `0xb..f` are still illegal. I'll update the spec to make this clear again for the next release.

Thanks again for catching this!

Cheers,
Ben

ben-marshall, 2021-06-14:
@JamesKenneyImperas  - See [this commit](https://github.com/riscv/riscv-crypto/commit/189808392853002e399a0412056f79904db96989) - hopefully this specifies things as they should be.

commit 18980839 (2021-06-14): spec: scalar: Fix #97 - missing constraints on aes64ks1i rnum

 On branch dev/next-release
 Your branch is up-to-date with 'origin/dev/next-release'.

 Changes to be committed:
	modified:   insns/aes64ks1i.adoc

 Changes not staged for commit:
	modified:   riscv-crypto-scalar-appx-rationale.adoc
	modified:   ../../extern/riscv-gnu-toolchain (modified content)
	modified:   ../../extern/sail-riscv (modified content, untracked content)
closed 2021-06-23 by ben-marshall