riscv/riscv-crypto #98: packu[h] -> packu[w]?

JamesKenneyImperas, 2021-06-14:
The change notes for 0.9.2 refer twice to `packu[h]` instructions. I don't think there is any such instruction `packuh` - do you mean `packuw`?

Thanks.

JamesKenneyImperas, 2021-06-14:
(Also, can you clarify the status of packu* and rev8.w? The change log says they are removed, but they are still in the specification.)

ben-marshall, 2021-06-14:
> The change notes for 0.9.2 refer twice to packu[h] instructions. I don't think there is any such instruction packuh - do you mean packuw?

Yes, that should have been `packuw`

> (Also, can you clarify the status of packu* and rev8.w? The change log says they are removed, but they are still in the specification.)

Gah. This release needs a complete do over, I'm sorry about this.

So, yes, these instructions _should have been removed_. Clearly I missed some (all?!) references to them in the spec.

I'll fix this up asap.

Cheers,
Ben

ben-marshall, 2021-06-14:
@JamesKenneyImperas  - Where in the spec are you seeing `rev8.w` and `packu*`? I can only see them in the appendix and in the `Zkt` section, and am now double checking I've not done something even more stupid than I thought?

ben-marshall, 2021-06-14:
Hopefully the commit above removes all unncessary mention of `packu[w]` and `rev8.w`. They _should_ still be mentioned in the `Zkt` section. I hope this is more clear? If not let me know.
Cheers,
Ben

JamesKenneyImperas, 2021-06-14:
Hi Ben - Ok, I saw them in the Zkt section (5.4.5) and misunderstood what that was all about - sorry.

closed 2021-08-06 by ben-marshall