riscv/riscv-crypto #100: mseccfg/mseccnf clarification (4.6)

JamesKenneyImperas, 2021-06-15:
Section 4.6 talks about `mseccfg` and also `mseccnf`, which is a CSR at address `0x390`. can you please clarify:

1. Are these the same thing?
2. Is the CSR are 0x390 only present if Zkr is implemented?
3. Is there any documentation of the remaining bits in `mseccnf`? I can't find any other details about it.

Thanks.

ben-marshall, 2021-06-15:
Hi @JamesKenneyImperas 

Yes, this is a typo, it should have read `mseccfg`.

The `mseccfg` register is defined / owned by the TEE task group. They let us have a single bit in it because it seemed like a logical place to put the `SKES` bit. I realise now the spec could do with expanding on what this register is and when it exists. Something like the following:

> `mseccfg` exists if `Zkr` is implemented, or if any of the "additional functionalities from the TEE group" (I'll describe these better later) are implemented. If `Zkr` is _not_ implemented, the `SKES` bit must be hardwired to zero. If `Zkr` is implemented and "the TEE functionality" is not implemented, all bits other than `SKES` must be hardwired to zero, or otherwise set according to the TEE specifications for a not implemented feature.

Would this have answered your above question? Baring exactly what the other bits do?

Cheers,
Ben

JamesKenneyImperas, 2021-06-15:
Hi Ben,

Yes, that would have helped. Thanks!

ben-marshall, 2021-06-15:
Great, I'll add that in for the next release.

closed 2021-06-23 by ben-marshall

mickflemm, 2021-07-05:
We switched mseccfg address to 0x747 as requested by Opcode and Consistency review.

commit dd917527 (2021-06-23): Spec: scalar: Fix #100 - mseccfg clarifications.

Add information on how/where/why the mseccfg CSR is implemented.

 On branch dev/next-release
 Your branch is up-to-date with 'origin/dev/next-release'.

 Changes to be committed:
	modified:   riscv-crypto-scalar-entropy-source.adoc

 Changes not staged for commit:
	modified:   ../../extern/riscv-gnu-toolchain (modified content)
	modified:   ../../extern/sail-riscv (modified content, untracked content)