riscv/riscv-crypto #101: Review considerations

JamesKenneyImperas, 2021-06-15:
I thought it might be helpful to point out a couple of unusual things about this extension that perhaps ought to be considered as part of the review. I'm not saying these are good or bad from a hardware perspective - I just want to make sure reviewers who might know much more about affected areas than I do are aware of them.

**1. Conflicting decodes**
The extension has instruction decodes that conflict and are mode-dependent. Specifically, the RV32 `aes32*` instructions all conflict with `aes64*` instructions which operate differently; this is unusual, and I haven't seen it in any other extension. The natural way to do this might have been to have separate decodes for `aes32*` and `aes64*` instructions, allowing `aes32*` instructions to be used in RV64 state too, but of course this would take more encoding space. I don't know what the hardware or tool implications of this might be, but I could imagine that it might complicate the workings of objdump and perhaps debuggers.

**2. Shared instructions with B-extension**
The extension shares subsets of instructions with the B extension at a very fine grain. For example, the grevi instruction is shared, but only with constants 7 (`rev.b`), 24 in RV32 mode only (`rev8`) and 56 in RV64 mode only (also `rev8`). This makes me worry that this might be difficult to implement in systems when both B and K are separately-enabled in the `misa` register: the hardware will have to know that when K is enabled and B is not, that only these specific constants are allowed in these specific modes, but if B is enabled (with or without K) then all constant values are legal.

Thanks.

ben-marshall, 2021-06-23:
I think these are good points which @aswaterman and others will consider during the architectural review.

About 1 in particular - this is another legacy of the crypto TG's time under the impression opcode space was much more scarce than it actually is. So that's how it came to be. As to whether it should be changed: I leave this to our reviewers.

aswaterman, 2021-06-23:
Note the C extension already reuses opcodes for different purposes for RV32 vs. RV64, e.g., C.FLW vs. C.LD.  It is annoying but it is a bridge we have already crossed in the 16-bit encoding space.  (I’m not yet taking a position on the merits of continuing that strategy for the AES instructions.)

ben-marshall, 2021-07-23:
Following up on this - after getting the arch-review feedback and discussing it at the 22/Jul/21 TG meeting, we decided _not_ to overlap the aes32 and aes64 opcodes.

You can see riscv/riscv-opcodes#77 for the exact changes. It's going in to riscv-opcodes ahead of the other spec updates to make sail/spike/binutils changes easier.

kdockser, 2022-10-29:
This was resolved as a part of Scalar Crypto review.

closed 2022-10-29 by kdockser