riscv/riscv-crypto #102: Annotate opcode type under encodings?

ksyx, 2021-06-17:
As in B extension, adding this annotation could save time locating them in the RISC-V spec table.

[image]

ben-marshall, 2021-06-21:
I agree this is a nice to have. I don't know how Bitmanip generated these diagrams. I used the script in `bin/better-parse-opcodes.py`. So some modifications to that might do the trick. But, I certainly won't get to it for a while. This is very much a cosmetic change so not high priority (for me). Eventually I'm sure all RISC-V specs will have this, but it might not make the public review version of scalar crypto.

kdockser, 2022-10-29:
This is a comment for the Scalar Crypto spec. Such changes are now the in domain of integrating specifications and making them consistent.

closed 2022-10-29 by kdockser