riscv/riscv-crypto #104: Spike illegal instruction was executed

enqnwhgdk317, 2021-06-21:
Hello. I want to use spike ISS for rv64k(scalar cryptography) extension.
But scalar crypto instructions are not executed in spike, printing "An illegal instruction was executed' .

<stdin>:25.39-29.9: Warning (interrupt_provider): /cpus/cpu@0/interrupt-controller: Missing #address-cells in interrupt provider
<stdin>:25.39-29.9: Warning (interrupt_provider): /cpus/cpu@0/interrupt-controller: Missing #address-cells in interrupt provider
bbl loader
z  0000000000000000 ra 0000000000000000 sp 0000003ffffffb50 gp 0000000000000000
tp 0000000000000000 t0 0000000000000000 t1 0000000000000000 t2 0000000000000000
s0 0000000000000000 s1 0000000000000002 a0 0000000000000000 a1 0000000000000000
a2 0000000000000000 a3 0000000000000000 a4 0000000000000000 a5 0000000000000000
a6 0000000000000000 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000003 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 000000000001007c va/inst 00000000b1448ab3 sr 8000000200006020
An illegal instruction was executed!

Also, there's no 
{REPO_HOME}/tool/pk-apply.sh  and 
{REPO_HOME}/tool/spike-apply.sh

Can I get some advice?
Thank you.

ben-marshall, 2021-06-21:
Hi there

I can't help without more information:
- What was the exact command line you used to run Spike?
- Which version of the toolchain did you use to compile the program?
- Which version of Spike are you using? Where did you get it from and when?
- Can you run any of the benchmarks included in the repository?

About the missing scripts - yes, it looks like the documentation is out of date there. I'll fix that when I get the chance.

Cheers,
Ben

enqnwhgdk317, 2021-06-24:
Hello.

1. I used
$ riscv64-unknown-elf-gcc -c assembly1.s
$ riscv64-unknown-elf-ld assembly1.o
$ spike pk a.out

and result shows
parallels@parallels:~/spike_simul$ spike pk a.out
bbl loader
z  0000000000000000 ra 0000000000000000 sp 0000003ffffffb50 gp 0000000000000000
tp 0000000000000000 t0 0000000000000000 t1 0000000000000000 t2 0000000000000000
s0 0000000000000000 s1 0000000000000002 a0 0000000000000000 a1 0000000000000000
a2 0000000000000000 a3 0000000000000000 a4 0000000000000000 a5 0000000000000000
a6 0000000000000000 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000003 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 000000000001007c va/inst 0000000031691693 sr 8000000200006020
An illegal instruction was executed!

and assembly1.s includes
---------------------------------------------------
.global _start

_start: addi s1, s1, 2
        addi s4, s4, 3
        aes64ks1i x13, x18, 6
        sha512sum1 x2, x15
        aes64es x12, x8, x3
        sha512sig0 x15, x15
        aes64dsm x12, x26, x12
        aes64dsm x10, x20, x12
        aes64ks1i x20, x23, 5
---------------------------------------------------------

and $ riscv64-unknown-elf-objdump -d a.out returns
-----------------------------------------------------------
a.out:     file format elf64-littleriscv

Disassembly of section .text:

0000000000010078 <_start>:
   10078:	0489                	addi	s1,s1,2
   1007a:	0a0d                	addi	s4,s4,3
   1007c:	31691693          	aes64ks1i	a3,s2,0x6
   10080:	10579113          	sha512sum1	sp,a5
   10084:	32340633          	aes32esi	a2,s0,gp,0x0
   10088:	10679793          	sha512sig0	a5,a5
   1008c:	3ecd0633          	aes32dsmi	a2,s10,a2,0x0
   10090:	3eca0533          	aes32dsmi	a0,s4,a2,0x0
   10094:	315b9a13          	aes64ks1i	s4,s7,0x5
------------------------------------------------------------

It seems addi instructions are executed well, but K-extension instructions are not being recognized.

2. tool chain version is 10.2.0 (I reinstalled riscv-gnu-toolchain and spike today)

parallels@parallels:~/spike_simul$ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 10.2.0

3. spike version is 1.0.1
 
parallels@parallels:~/spike_simul$ spike
Spike RISC-V ISA Simulator 1.0.1-dev

4. I tried to run aes benchmark but it didn't work.
aes_128_ks.s:9: Error: junk at end of line, first unrecognized character is `/'
aes_128_ks.s:10: Error: junk at end of line, first unrecognized character is `/'
aes_128_ks.s:20: Error: junk at end of line, first unrecognized character is `/'
aes_128_ks.s:21: Error: junk at end of line, first unrecognized character is `/'
aes_128_ks.s:30: Error: junk at end of line, first unrecognized character is `/'
aes_128_ks.s:31: Error: unrecognized opcode `aes_load_state RK_LO,RK_HI,CK,TMP1,TMP2'
aes_128_ks.s:33: Error: unrecognized opcode `aes_dump_state RK_LO,RK_HI,RK,TMP1,TMP2,0*16'
aes_128_ks.s:35: Error: illegal operands `aes64ks1i TMP1,RK_HI,0'
aes_128_ks.s:35: Error: illegal operands `aes64ks2 RK_LO,TMP1,RK_LO'
aes_128_ks.s:35: Error: illegal operands `aes64ks2 RK_HI,RK_LO,RK_HI'
There are other similar errors in the benchmark.

Do I have to do something additionally to simulate K-extension in spike??

enqnwhgdk317, 2021-06-28:
Hello!

$ spike --isa=rv64gck pk a.out

This works well. 
Thank you!

closed 2021-06-28 by enqnwhgdk317