riscv/riscv-crypto #106: proposal to extend OPST field

jnk0le, 2021-07-23:
There are cases when entropy source is not configured (as initial state after boot) or misconfigured (e.g. by some weird race condition, like [STM32F4 errata 2.1.13](https://www.st.com/resource/en/errata_sheet/es0182-stm32f405407xx-and-stm32f415417xx-device-limitations-stmicroelectronics.pdf) vs flash waitstates/cache)

Those will result in infinite BIST as that's the only way it can be currently implemented (no recovery from DEAD)
That might also be the source of very hard to debug bugs.

Instead I'd propose the addition of `UNAVAIL` state that allows quick detection and logging any kind of "never happens" issues.

Similarly to DEAD, there is no need to poll after UNAVAIL as it won't fix itself.

Assuming the arch review proposed codes:
>00 = BIST, 01 = WAIT, 10 = ES16, 11 = DEAD

those could be expanded to e.g.
```
000 = BIST, 001 = WAIT,
100 = ES16,
110 = UNAVAIL, 111 = DEAD
```
It can be handled very similarly to what is currently proposed
- BIST/WAIT have cleared MSB (test with BGE{Z})
- DEAD/UNAVAIL can be treated similarly to BIST/WAIT (and current DEAD) by checking only middle bit (left shift + BGEZ)

Also
Earlier versions of document had following text (no more in 0.9.3):
>Implementations do not need to implement DEAD as it may not require an end-user notification

With UNAVAIL state, the peripherals could have explicitly configurable tamper protection(DEAD) and utilize UNAVAIL state for requiring software intervention. Not sure how it will stand in certification.

jnk0le, 2021-07-24:
"NOISE_TEST" should also fall into UNAVAIL.

mjosaarinen, 2021-07-24:
Hi,

I don't think that a generic driver component would know what to do with the UNAVAIL information. Couldn't one just use the custom bits sentropy[23:16] to signal this condition if there is a vendor driver that can do something about it?

Anyway, being a key security component, a non-configured or misconfigured entropy source probably should be considered DEAD ?

Cheers,
- markku

jnk0le, 2021-07-25:
>I don't think that a generic driver component would know what to do with the UNAVAIL

Should treat similarly to DEAD. 

>Couldn't one just use the custom bits sentropy[23:16] to signal this condition if there is a vendor driver that can do something about it?

Probably, but custom == fragmentation or establishing "de-facto standards".
UNAVAIL is more about signalling that further polling won't give any more entropy. Generic software can immediately react instead of infinitely waiting on BIST.

>Anyway, being a key security component, a non-configured or misconfigured entropy source probably should be considered DEAD ?

In this case we are back to Schrödinger's cat problem.

mjosaarinen, 2021-07-28:
Well, if the discovery yields a Zkr, which is subsequently DEAD, isn't that the same as signaling that it's unavailable/broken?  DEAD means that the source is unavailable, not that it is not implemented.

jnk0le, 2021-08-11:
Discovery mechanism is a non goal of UNAVAIL.

>Well, if the discovery yields a Zkr, which is subsequently DEAD

Is it currently allowed to undie if "discovery yields a Zkr, which is subsequently DEAD" and then entropy source gets initialized (assuming it wasn't polled)?

mjosaarinen, 2021-08-16:
Things shouldn't undie, but if the entropy source module is uninitialized, shouldn't it be in BIST mode?  BIST is typically the mode where the source should be immediately after start -- even before it can even be DEAD (after failing the the said Built-In Self Test). If the source doesn't have a BIST, it can't really be a 90B module, and really has no business being interfaced through the entropy source interface.

mjosaarinen, 2021-08-16:
I closed this but reopened it in the hopes that the proposer can explain why BIST can't be used for configuration here.

Alternatively, a clear definition, behavioral model, and mapping to SP 800-90B should be given.

jnk0le, 2021-08-30:
SP 800-90B doesn't explicitly prohibit any states before BIST, it only mandates startup BIST before outputting entropy. 
So the: `UNAVAIL -> BIST -> WAIT/ES16` transitions seem to be lagal in my understanding of SP800-90B.

If the core/soc implements self-starting entropy source etc. then it doesn't need to implement UNAVAIL.

It is possible to implement it equivalantly as BIST with custom bit set, but it lacks the property of "treat as DEAD" without extra branching as well as standardization.

mjosaarinen, 2021-09-03:
It seems that there are no important implementation issues with the current interface and there is no clear motivation for the change from a standards perspective.

closed 2021-09-03 by mjosaarinen