riscv/riscv-crypto #114: aes32*si and sm4* index into rs2, unlike other RISC-V instructions

kasanovic, 2021-08-23:
aes32*si and sm4* instructions index bytes from rs2 whereas other shift and xperm
instructions index bytes from rs1. Swapping use of two register fields
would probably reduce datapath muxing costs.

mjosaarinen, 2021-08-23:
The way these instructions are used, they almost always rewrite back to rs1 register (that's why there was a compressed encoding for it). Swapping the source registers around would essentially always have rd=rs2, which is much less common in compiler-generated code (even if C extension is not used), and hence less of a timing target in hardware implementations.

derekatkins, 2021-08-23:
@mjosaarinen --  You say:
> hence less of a timing target in hardware implementations.

Could you clarify which direction is "less of a timing target"?  Maybe I need more tea, but I'm not sure I follow which way you are leaning.

Thanks.

mjosaarinen, 2021-08-23:
Hi,

Flipping rs1 and rs2 would be that rs2 would very often be sourced from the result of the immediately previous op, which is not as common as rd=rs1 overwrite. Compressed instructions are effectively rd=rs1 instructions, etc. One can imagine that some simple pipeline implementation could even impose a cycle penalty on sequences with rd=rs2? 

Recall that AES instructions are primarily used to implement AES and hence we essentially know how they are used; it's "always"  rd=rs1 currently. The change would make all of this rd=rs2. (e.g. https://github.com/riscv/riscv-crypto/blob/f0bb63e65347ea696c2fbbd1f511e5b4ec8d0324/benchmarks/aes/zscrypto_rv32/aes_enc.S )
```
        aes32esmi   a4, a4, t0, 0                   // Even Round (replaced
        aes32esmi   a4, a4, t1, 1
        aes32esmi   a4, a4, t2, 2
        aes32esmi   a4, a4, t3, 3

        aes32esmi   a5, a5, t1, 0
        aes32esmi   a5, a5, t2, 1
        ...
```

Cheers,
- markku

> @mjosaarinen -- You say:
> 
> > hence less of a timing target in hardware implementations.
> 
> Could you clarify which direction is "less of a timing target"? Maybe I need more tea, but I'm not sure I follow which way you are leaning.
> 
> Thanks.

jnk0le, 2021-08-24:
>it's "always" rd=rs1 currently

In case of rolled-in mov-less implementation rd!=rs1 takes place in 4 out of 16 aes32 instructions

e.g. (not tested)
```
.aes_enc_block_l0:
    
	lw      U0, 16(RK)
	lw      U1, 20(RK)
	lw      U2, 24(RK)
	lw      U3, 28(RK)

	// empty T0
	aes32esmi   U1, U1, T0, 3
	aes32esmi   U2, U2, T0, 2
	aes32esmi   U3, U3, T0, 1
		
	aes32esmi   U0, U0, T1, 1
	aes32esmi   U0, U0, T2, 2
	aes32esmi   U0, U0, T3, 3 
	aes32esmi   T0, U0, T0, 0 // col 0

	//empty T1
	aes32esmi   U2, U2, T1, 3
	aes32esmi   U3, U3, T1, 2
		
	aes32esmi   U1, U1, T2, 1		
	aes32esmi   U1, U1, T3, 2
	aes32esmi   T1, U1, T1, 0  // col 1
		
	//empty T2
	aes32esmi   U3, U3, T2, 3
		
	aes32esmi   U2, U2, T3, 1
	aes32esmi   T2, U2, T2, 0  // col 2
		
	aes32esmi   T3, U3, T3, 0  // col 3
	
	addi    RK, RK, 16
	bne     RK, KP, .aes_enc_block_l0
```
it can be further optimized to avoid sourcing operand from previous instruction rd

kasanovic, 2021-08-24:
I'm mystified by the "timing target" comment - none of these options causes data-dependent timing.

Whether rd=rs1 or rd=rs2 is irrelevant for processor performance- even simple realistic in-order pipelines will have full bypassing.  Code scheduling is a good optimization, but is unaffected by choice of rs1/rs2.

These are clearly not candidates for compressed instruction encodings, so following rd=rs1 pattern there is moot point.

Swapping rs1/rs2 to enable reuse of existing shifter with rs1 as data input is admittedly only a relatively small improvement, but doesn't impact anything else (except late-change effect on spec development effort).

ben-marshall, 2021-08-24:
> Swapping rs1/rs2 to enable reuse of existing shifter with rs1 as data input is admittedly only a relatively small improvement, but doesn't impact anything else (except late-change effect on spec development effort).

I don't think this would be a good idea in an implementation, though simply _enabling_ it doesn't hurt.

The size and depth of the circuit for implementing these instructions is dominated by the AES SBOX. Re-using an existing barrel (I assume) shifter just to extract a byte would likely make both circuits worse than they could be by splitting the logic. The initial "extract byte" part of these instructions is tiny by comparison. This would likely only show up post P&R though. If you just looked at synthesis it would look like an improvement.

mjosaarinen, 2021-08-24:
@kasanovic 
> I'm mystified by the "timing target" comment - none of these options causes data-dependent timing.

Should have been clearer -- that was about timing closure, not timing attacks.

It is true that even if the selection of fixed things like rd=rs1 vs rd=rs2 affects execution latency, that is not a timing attack as long as both forms have _data_-independent latency. So that is not a worry..

> Whether rd=rs1 or rd=rs2 is irrelevant for processor performance- even simple realistic in-order pipelines will have full bypassing. Code scheduling is a good optimization, but is unaffected by [...]

We did have an opcode-space optimization based on this, but obviously not compressed encoding .. indeed too many registers are used in implementations to benefit from compressed forms and the instruction is too rare.

> Swapping rs1/rs2 to enable reuse of existing shifter with rs1 as data input is admittedly only a relatively small improvement, but doesn't impact anything else (except late-change effect on spec development effort).

As a practical implementation technique that will lead to a rather large circuit depth (thanks to the S-Box), so in practice, one still wants to do a byte select mux rather than reuse a few stages of a barrel shifter. 

Cheers,
- markku

kasanovic, 2021-08-25:
To be clear, I am not opposed to sticking with current scheme as this is a minor point, but I understood that the encodings had to shift after arch review anyway and this seemed liked a marginal improvement in implementation options and consistency at this point before things freeze. 

Regarding barrel shifters - this terminology was originally used for single-cycle (combinational) shifters as opposed to ones that took multiple cycles to shift, and also a particular crossbar circuit implementation style.
Modern static logic designs will build combinational shifters using multiple mux stages (usually radix-2 or radix-4).  The byte-level extract from 32b represents the most expensive stages (shift by 16,8)  of the shifter (the remaining shifts by 4,2,1 bits are cheaper due to less wiring).  Tapping that output (so not adding remaining shift stages to critical path) to reduce wire/muxes could be desirable area/latency tradeoff in some implementations.  Of course, implementer is free to build dedicated circuit also.  Current scheme mandates adding a separate byte mux on the rs2 port.

jnk0le, 2021-08-30:
Those instructions need to rotate pre-xor result. So if the logic gets reused here, is there anything else that can be reused for input byte mux?