riscv/riscv-crypto #120: OPST position on rv64

jnk0le, 2021-09-02:
Should it be always at [31:30] positions?
Having it at [XLEN-1:XLEN-2] gives a minor saving in polling code, that can be reused for RV32 and RV64.

mjosaarinen, 2021-09-03:
Can you give an example of how that is a saving, and also a quantitative argument why it is important? Also: RV32 and RV64 would need different codes in that case, so I don't understand your reuse argument. Currently, they can use exactly the same code which is helpful for Linux kernel, etc.

Note that only a system service or kernel should implement polling on the CSR. There are not many instances of it, ideally only 1 per system.  Furthermore, it is an opportunistically used, low-bandwidth interface.

jnk0le, 2021-09-03:
>Can you give an example of how that is a saving
Just one constant load for bge.

Very minor improvement, sign extending 32bit value would also have similar effect.
However "set to zero" bits can be converted to custom at any moment.

jnk0le, 2021-09-03:
I see the spec is already frozen and that's not significant change.

closed 2021-09-03 by jnk0le