riscv/riscv-crypto #125: Zero and sign extension of pack.h vs pack.w

RogierBrussee, 2021-09-04:
The pack.h instruction _zero_ extends while the pack.w instruction _sign_ extends. 
Because the instructions are simple and otherwise very similar, maybe for ease of reference this should be mentioned in the synopsis (it is already mentioned in the description). 

Arguably it would be less confusing if pack.h would be renamed pack.hu

(Same issue as #160 on the bitmanip GitHub)

ben-marshall, 2021-09-08:
I'm happy adding a note to the synopsis to each instruction if that'd make things clearer. These instructions & their names were originally defined by the bitmanip task group so we are inheriting any oddities like this.

(riscv/riscv-bitmanip#160)

RogierBrussee, 2021-09-09:
Thanks Ben!

As for the naming: Bitmanip has proved to be a slow process and it seems
that it is going to be sliced up in different Xb* extensions.  The Zbk
cryptography extension seems the first to use the PACK.* instructions so
you may be able to set the definitive name.

But I raised the same issue at bitmanip as well.

Incidentally, the more precise naming does suggest  instructions PACK.H
(which packs bytes and  sign extend) and PACK.WU (which packs halfwords and
zero extends). Whether they are worth it, not to mention a good idea at
this stage in the process, is entirely another matter of course. I think
the current instructions come from the convention of having  uint16_t
 represented by a zero extended halfword but uint32_t  being represented as
sign extended.  I _think_ that Claire Wolfe also checked that a struct foo
{uint16_t first; uint16_t second} or struct RGBA{uint8_t R,G,B, A} can be
constructed in a register using pack and that it is OK that these values
are sign extended because upper 32 bits are ignored in the calling
convention.

Rogier

kdockser, 2022-10-29:
This comment was for the Scalar Crypto extension which has been finalized and ratified. No changes of this type may be made at this time.

closed 2022-10-29 by kdockser

RogierBrussee, 2022-10-31:
Yes this ship has sailed. But I made the comment during the ratification
phase.

Thanx

kdockser, 2022-11-01:
Sorry, I didn't mean to imply that your comment was late. I was in the process of cleaning out old issues and was just concentrating on whether or not there was something that we need to do at the current time.

Thanks,
ken