riscv/riscv-crypto #128: rv32 sha512sigDX wrong order

marcfedorow, 2021-09-20:
Note to software developers says:
```
The  entire  Sigma0  transform  for  SHA2-512  may  be  computed  on  RV32  using  the  following
instruction sequence:
sha512sig0l    t0, a0, a1
sha512sig0h    t1, a0, a1
```
This is not true. Actually sigma0  transform  for  SHA2-512  may  be  computed  on  RV32  using  the  following
```
sha512sig0l    t0, a0, a1
sha512sig0h    t1, a1, a0
```
It is obviously a typo whilst this instruction is `ror`-based (as `sha512sum0r`).

marcfedorow, 2021-09-20:
Also note that OpenSSL marks `Sum` functions as `Sigma`, and `Sigma` functions as `sigma`.
https://github.com/openssl/openssl/blob/master/crypto/sha/sha512.c
It is probably better to use OpenSSL's naming, but I am not sure.

ben-marshall, 2021-09-20:
Nice catch! Thank you, I will update this in the spec when I have some time on Wednesday.

ben-marshall, 2021-09-22:
Hi @marcfedorow - I've updated the spec in 1b66e0a - it turns out several instructions had the same typo. Thanks again for spotting this, I'll leave the issue open in case others spot it to avoid duplicates.
Cheers,
Ben

closed 2022-10-29 by kdockser