riscv/riscv-crypto #130: Scalar Cryptography v1.0.0-rc2. Incorrect ZIP operarion description.

bulat242, 2021-09-21:
Chapter 3 -> 3.49 (Page 58):
**Operation**
foreach (i from 0 to xlen/2-1) {
X(rd)[2*i] = X(rs1)[i]
X(rd)[2*i+1] = X(rs1)[i+xlen/2]
}
This description pass to UNZIP.

Correct description of ZIP:
**Operation**
foreach (i from 0 to xlen/2-1) {
X(rd)[i] = X(rs1)[2*i]
X(rd)[i+xlen/2] = X(rs1)[2*i+1]
}

ben-marshall, 2021-09-22:
Hi @bulat242 

Thanks for spotting this! Just to be clear - it looks like the pseudocode for `zip` and `unzip` are the wrong way around, right?

I'll have to do some research on how to handle this, since although it's very simple to swap the descriptions in the spec, because its frozen, there might be other implications.

Cheers,
Ben

bulat242, 2021-09-22:
Yes, the descriptions for zip and unzip have to be swapped in a general sense. (UNZIP operation in spec is described as "inverse zip").

ben-marshall, 2021-09-22:
> (UNZIP operation in spec is described as "inverse zip").

Yeah, that's a rubbish description for sure. I'll fix that.

Okay, I'll follow this up at the weekly CETG meeting tomorrow and ask how to handle it.

bulat242, 2021-09-22:
And I would like to clarify about **rev8** and **brev8** descriptions. 
Are these descriptions same?
rev8:
foreach (i from 0 to (xlen - 8) by 8)
brev8:
foreach (i from 0 to sizeof(xlen) by 8)

ben-marshall, 2021-09-22:
> And I would like to clarify about rev8 and brev8 descriptions.

The _intent_ is to step through every byte in the register, so yes the `foreach` statements are supposed to be equivalent. It makes no sense to express the same thing two different ways, so I'll make sure we use something consistent in the final versions.

bulat242, 2021-09-22:
Okey, thank you for explanation!

ben-marshall, 2021-09-24:
@bulat242 - The plan for fixing this is to make the necessary changes in the spec and do a `v1.0.0-rc3` release, and publicise this on the isa-dev mailing list. I'm double and triple checking things at the moment, but hopefully it'll be out if not today then early next week.

Out of curiosity - how did you catch the problem?

bulat242, 2021-09-24:
@ben-marshall Okey :)
I compared these descriptions of the instructions with old descriptions from bitmanip spec (v0.93).

ben-marshall, 2021-09-24:
@bulat242  - [Here's the fix](https://github.com/ben-marshall/riscv-bitmanip/commit/bdf5d148878c90ff9e21b06acaa50f28472176ba), hopefully this is a bit clearer than before!

ben-marshall, 2021-09-24:
I'm going to leave the issue open for now in case other people spot the same problem and try to report it. Thanks again for the catch!

bulat242, 2021-09-24:
Okay, you are welcome.

bulat242, 2021-09-28:
@ben-marshall Hi!
I have a new question about encoding of _**zip/unzip**_ insns. 
In current scalar cryptography spec ZIP and UNZIP insns have different values in 20th bit. I understand the 20th bit in Bitmanip spec (v. 0.93) was included in shamt field of shfli/unshfli insns and was set for ZIP/UNZIP cases.
Was ZIP/UNZIP insns encoding in current crypto spec derived from bitmanip spec (v 0.93) and is it correct?

ben-marshall, 2021-09-29:
Hi @bulat242 - The encodings were not modified from the proposed Bitmanip ones. If they are different that's likely a mistake. Note that we are only standardising the RV32 versions of zip/unzip. I _think_ the RV64 versions indeed had a different length shamt field. Could that be where they difference came from? I think the 0.93 Bitmanip spec in table 2.7 only gave RV64 listings for the SHAMT field.

marcfedorow, 2021-09-29:
Hi @ben-marshall
Actually 0.9.3 butmanip spec gave RV128 encodings — you can easily assure by checking 7-bit immediate for SLLIU.W (RV32 and RV64 require 5 and 6 bits respectively).
Due to this reason (and explicit placing of "inv" bit to the generalization table) someone might have mistakenly decided that "inv" bit is a part of immediate.
This is not true tho — immediatw only contains shift amount (shamt), inv bit is 14th one.
Thus both zip and unzip have to have `01111` bits in RS2 position (RV32 only! `11111` for RV64).

bulat242, 2021-09-29:
Yes, the encoding of **_zip/unzip_** (rv32 only) insns should look like below:
**_zip_**    : 0000100_01111_?????_001_?????_0010011
**_unzip_**: 0000100_01111_?????_101_?????_0010011

ben-marshall, 2021-09-29:
Ah yes I see now. It looks like bits 24:20 in the scalar crypto spec were taken from the wrong part of the tables in Bitmanip 0.93. I'll fix this in the `rc3` release so that for both, bits 24:20 are `01111` as you say.
Thanks again and good catch!
Cheers,
Ben

bulat242, 2021-09-29:
Okey, happy to help!

ben-marshall, 2021-10-01:
Hi @bulat242 , @marcfedorow - See [here](https://github.com/ben-marshall/riscv-bitmanip/commit/bc3a9173df2ab0f5704d450f779eb17547fbc74b) for the fix to the immediates. I'll do a release later today and make sure it gets publicised on isa-dev. Thanks for all your help with this!

marcfedorow, 2021-10-01:
Hi @ben-marshall 
Thank you for ping!
Current descriptions seem to be incorrect tho.
shfl (zip) should do 3, 2, 1, 0 stages.
unshfl (unzip) should do 0, 1, 2, 3 stages.
This could be mixed up because B-ext 0.9.3. graphs should be read upwards (from down to up) -- check C code or other graphs for cross-check.

bulat242, 2021-10-01:
Yes, the descriptions of zip/unzip in crypto spec v1.0.0-rc2 were correct.
I'm sorry.

marcfedorow, 2021-10-05:
@ben-marshall please note that this issue is not resolved yet

ben-marshall, 2021-10-05:
Hi @marcfedorow - yes it's on my list to fix (again). I'll get to it tomorrow.

bulat242, 2021-10-06:
Hi, @ben-marshall !
You are right. The encodings now (in spec rc3) are correct, but unfortunately (because of my inattention) the pseudo-code of zip/unzip is incorrect.

closed 2022-10-29 by kdockser