riscv/riscv-crypto #132: name consistency of brev8/rev8 and xperm8/xperm4

Rolfy360, 2021-10-01:
In the A.4, all bitmanip instructions are reviewed again,
but there is some name inconsistency:

A.4.2
rev.b <- this is called brev8 in Zbkb
rev8

BTW, the instruction names of these 2 are confusing,
I'll say I love bitrev/rbit/bswap more.
(There is a rule of naming grevi in old B-ext draft.)

A.4.6
xperm.n <- xperm4 (Zbkx)
xperm.b <- xperm8 (Zbkx)

ben-marshall, 2021-10-01:
Hi @Rolfy360

You're absolutely right, the names for some instructions have changed over time and while things have been kept up to date in the normative spec, the appendix has been missed. I'll update that in the next RC.

> BTW, the instruction names of these 2 are confusing,

Personally I like the longer names you've picked, but `rev8` etc were chosen for us by others, so they're a bit out of our control unfortunately.

Cheers,
Ben

ben-marshall, 2021-10-01:
@Rolfy360  - See [here](https://github.com/riscv/riscv-crypto/commit/16e7b5273346ff11f05725ec80106ad7c58409f6) for the changes to address this.

marcfedorow, 2021-10-04:
Hi, @ben-marshall 
As I mentioned [here](https://github.com/riscv/riscv-crypto/issues/115#issuecomment-911662597), brev8 is a very confusing name because it is inconsistent with the old B-ext draft naming scheme.
It could be much more clear if K-spec either takes this naming as is or names instructions with new names.

ben-marshall, 2021-10-04:
Hi @marcfedorow - I'll bring this up at the next meeting on Thursday for sure, I agree it's important to keep things consistent. Unfortunately the task group didn't get to choose the names for these instructions, we were told "please implement these" like you saw in the thread for #115 .

marcfedorow, 2021-10-06:
@ben-marshall May I participate in this meeting? How could I do it?

kdockser, 2022-10-29:
This was for the Scalar Crypto spec was has been finalized and ratified. There is nothing that needs to be done for the spec.

closed 2022-10-29 by kdockser