riscv/riscv-crypto #143: Consider add GM/T 0005-2012 randomness test

luojia65, 2021-12-08:
### Concerns

GM/T 0005-2012 is a national standard in China for randomness test of commercial cryptography applications. In some regions it's required in law to use GM/T 0005 test method series to pass national standard verifications.

### Details

https://github.com/riscv/riscv-crypto/pull/142

mjosaarinen, 2021-12-08:
Isn't this a test that can be applied to pseudorandom output, not necessarily the entropy source? One can pass the national standard verifications that way perhaps.

kdockser, 2022-10-30:
The creation of randomness tests to pass national standards is beyond the scope of the RISC-V Crypto specifications and deliverables. Implementers will need to create such tests or obtain them from appropriate 3rd parties.

closed 2022-10-30 by kdockser