riscv/riscv-crypto #147: [RV64] encoding of "packw" will conflict with "zext.h" if rs2 field of "packw" is equal to zero

yichanfeng, 2021-12-27:
Hi,

I have a question about instruction encoding.  
=> Do we need to adjust instruction encoding of **packw** under **RV64** configuration because encoding of **packw** will conflict with **zext.h** if **rs2** field of **packw** is equal to zero ?

[RV64, B-extension] "zext.h" encoding

> [31:25]: 0000100, [24:20]: **00000,** [19:15]: rs field, [14:12]: 100, [11:7]: rd field, [6:0]: 0111011

[RV64, Scalar-Crypto] "packw" encoding

> [31:25]: 0000100, [24:20]: **rs2** field, [19:15]: rs1 field, [14:12]: 100, [11:7]: rd field, [6:0]: 0111011

Thanks!

ben-marshall, 2022-01-03:
Hi there - sorry about the long reply time, I've been on holiday over Christmas in the UK.

So, the `packw` encoding was re-used from the original one proposed by the bitmanip task group, so any encoding issues or rationale questions might be better asked there. I'll need to investigate properly when I have time, but this might have been deliberate because the operations overlap somehow. I.e. you could do `zext.h` and `packw` with the same underlying instruction, but just changing the register addresses.

yichanfeng, 2022-01-04:
Thank you for the reply.

This issue has been resolved in bitmanip task group (https://github.com/riscv/riscv-bitmanip/issues/170).

Please close this ticket.

closed 2022-01-04 by yichanfeng