riscv/riscv-crypto #148: EEW casting is broken due to vsetvl rules

jnk0le, 2022-01-07:
> This instruction treats `EEW=128`, regardless of `vtype.vsew`

according to paragraph 6.3 of RVV spec `vl` can be set to

>ceil(AVL / 2) ≤ vl ≤ VLMAX if AVL < (2 * VLMAX)

So if we want to stripmine 3 aes blocks with `VLEN == 256`, the uarch is allowed to split AVL equally to 192 bits (3*64bits) in both iterations due to better lane utilization. (spec doesn't specify behaviour in this situation yet, I'm assuming it will end up as illegal instruction ex)

The most straightforward and working approach I can think of, is to:
- require exact `SEW` (128 for aes)
- add "half operational state" (SEW>=128) where only loads, stores, bitwise, EDIV and dedicated insns are allowed 
- [OPT] EDIV (with sub elem masks for less reg pressure) allows whole CTR to be done with single vsetvli per iteration

without EDIV it brings issues with dirty casting (CTR incr, clmuls, etc.) though.

jnk0le, 2022-04-23:
>allows whole CTR to be done with single vsetvli per iteration

correction: with some kind of vgrev (vrev8) (big endian incr), otherwise it needs to do shift and add where vrgather might turn out to be faster

kasanovic, 2022-07-22:
I added https://github.com/riscv/riscv-v-spec/commit/44c9b212c0928c6e8fefeb652ceb794810cd01d9 to address this issue.

closed 2022-10-29 by kdockser

ebiggers, 2023-12-31:
Was this actually fixed?  The needed behavior is specified in the [RISC-V Vector Element Groups document](https://github.com/riscv/riscv-v-spec/blob/master/element_groups.adoc):

> When AVL > VLMAX, the value of vl must be set to either VLMAX or a positive integer multiple of EGSMAX.

However, that document is a "Concept - not part of any standard".  It is not clear that all CPUs will implement that behavior.

kdockser, 2024-01-02:
Thanks for pointing this out Eric.

The Vector Crypto Extensions places the needed requirement on _vl_ in _Section 1.5 Instruction Constraints_ :
`Since vl and vstart refer to elements, Vector Crypto instructions that use elements groups (See Section 1.4) require that these values are an integer multiple of the Element Group Size (EGS).`

So, yes, this is fixed.

It might have been helpful to have included a non-normative note in the Vector Crypto Specification to point out the implications on selecting `vl` when AVL < (2* VLMAX). However, since the RISC-V Vector Elements Group document (which is pointed to by the Vector Crypto Specification) provides such a _reminder_, I believe that we are well covered.

allenjbaum, 2024-01-02:
IT is covered, but non-normative notes added now will avoid lots of email
traffic explaining it later

ebiggers, 2024-01-18:
> It might have been helpful to have included a non-normative note in the Vector Crypto Specification to point out the implications on selecting vl when AVL < (2* VLMAX). However, since the RISC-V [...]

To be clear, you believe that software does not need to do anything special, because CPUs will take this reminder into account?  Or does software need to handle this, for example by executing `vsetvli rd, zero, ...` to explicitly set the vl to VLMAX, whenever the remaining length is greater than VLMAX?