riscv/riscv-crypto #149: Should c.srli, c.srai, and c.slli be part of Zkt on RV64?

topperc, 2022-01-17:
As far as I know these instructions are valid on RV64. Should they be part of Zkt? This appears to have been noted in the Arch Review Chapter 5 here https://wiki.riscv.org/download/attachments/350001/Scalar%20Crypto%20Arch%20Review%20Results.pdf

mjosaarinen, 2022-01-25:
Yes, they absolutely should be there. They were in the earliest proposals https://github.com/rvkrypto/riscv-zkt-list/blob/main/zkt-list.adoc but somehow got lost along the way.

( I think this has happened due to a request to remove the RV128 instructions C.SRLI64, C.SRAI64, and C.SLLI64 -- this was interpreted as removal of C.SRLI & co from RV64 somehow. )

ben-marshall, 2022-02-18:
The fix for this has now been released in [v1.0.1](https://github.com/riscv/riscv-crypto/releases/tag/v1.0.1-scalar).

closed 2022-02-28 by mjosaarinen