riscv/riscv-crypto #158: AES endianness

lvanderwiele, 2022-04-17:
Hi everyone. I'm currently writing my master's thesis on RVK and encountering a bit of an issue with AES.

If I understood the [paper on RVK AES](https://tches.iacr.org/index.php/TCHES/article/view/8729/8329) properly, the code required to run a full round of AES encoding is as follows

```
lw                  a0, 16(RK)//  Load  Round  Key
lw                  a1, 20(RK)
lw                  a2, 24(RK)
lw                  a3, 28(RK) // t0 ,t1 ,t2 ,t3  contains  current  round  state.

aes32esmi    a0 , a0, t0, 0 //  Next  state  for  column  0.
aes32esmi    a0 , a0, t1, 1 //  Current  column  0 in  t0.
aes32esmi    a0 , a0, t2, 2 //  Next  column  0  accumulates  in a0
aes32esmi    a0 , a0, t3, 3

aes32esmi    a1 , a1, t1, 0 //  Next  state  for  column  1.
aes32esmi    a1 , a1, t2, 1
aes32esmi    a1 , a1, t3, 2
aes32esmi    a1 , a1, t0, 3

aes32esmi    a2 , a2, t2, 0 //  Next  state  for  column  2.
aes32esmi    a2 , a2, t3, 1 
aes32esmi    a2 , a2, t0, 2
aes32esmi    a2 , a2, t1, 3

aes32esmi    a3 , a3, t3, 0 //  Next  state  for  column  3.
aes32esmi    a3 , a3, t0, 1
aes32esmi    a3 , a3, t1, 2
aes32esmi    a3 , a3, t2, 3 // a0 ,a1 ,a2 ,a3  contains  new  round  state

```
It would seem however that the code does not work as expected (at least not when comparing it to the example vectors found here: http://www.herongyang.com/Cryptography/AES-Example-Vector-of-AES-Encryption.html

I've gone through a few step to find the root of the issue so here's what I found.

Let us imagine the following 128-bit word: 0x000102030405060708090a0b0c0d0e0f. Using little-endian conventions, the first 32-bit subword should be 0x0c0d0e0f, the second 0x08090a0b and so forth with word thus being {w3, w2, w1, w0}. Similarly, the key is a 128-bit word expressed as {k3, k2, k1, k0}, the 32-bit word post ShiftRows and SubBytes is {s3, s2, s1, s0}, the result post MixColumns is {m3, m2, m1, m0} and  the cyphertext is {c3, c2, c1, c0}.

I tried to get the first 32 bits of the cyphertext by applying the 4 first four aes instructions mentioned above. However, after doing so (with MixColumns disabled and the key set to 0x0 to be able to examine the ShiftRows-SubBytes result), I noticed what I was getting wasn't s0 but s3 which would thus imply `lw a0, 16(RK)` is not loading the first 32 bits of the key (s0) but the last ones (s3).

Furthermore, applying the MixColumns operation yielded the wrong results and the coefficients had to be moved for it to work.

In the end, I got something working by doing this to my testbench (Verilog code):

```
aes32_2dir aes0(.rs1(key[96+:32]), .rs2(vector[0+:32]), .bs(2'd0), .mix(mix), .dec(1'b0), .rd(temp0));
aes32_2dir aes1(.rs1(temp0), .rs2(vector[32+:32]), .bs(2'd1), .mix(mix), .dec(1'b0), .rd(temp1));
aes32_2dir aes2(.rs1(temp1), .rs2(vector[64+:32]), .bs(2'd2), .mix(mix), .dec(1'b0), .rd(temp2));
aes32_2dir aes3(.rs1(temp2), .rs2(vector[96+:32]), .bs(2'd3), .mix(mix), .dec(1'b0), .rd(temp3));
```
Which ends up giving me c3 instead of c0. The rest of the code is as follows

aes32_2dir aes4(.rs1(key[64+:32]), .rs2(vector[0+:32]), .bs(2'd1), .mix(mix), .dec(1'b0), .rd(temp4));
aes32_2dir aes5(.rs1(temp4), .rs2(vector[32+:32]), .bs(2'd2), .mix(mix), .dec(1'b0), .rd(temp5));
aes32_2dir aes6(.rs1(temp5), .rs2(vector[64+:32]), .bs(2'd3), .mix(mix), .dec(1'b0), .rd(temp6));
aes32_2dir aes7(.rs1(temp6), .rs2(vector[96+:32]), .bs(2'd0), .mix(mix), .dec(1'b0), .rd(temp7));
	
aes32_2dir aes8(.rs1(key[32+:32]), .rs2(vector[0+:32]), .bs(2'd2), .mix(mix), .dec(1'b0), .rd(temp8));
aes32_2dir aes9(.rs1(temp8), .rs2(vector[32+:32]), .bs(2'd3), .mix(mix), .dec(1'b0), .rd(temp9));
aes32_2dir aesa(.rs1(temp9), .rs2(vector[64+:32]), .bs(2'd0), .mix(mix), .dec(1'b0), .rd(tempa));
aes32_2dir aesb(.rs1(tempa), .rs2(vector[96+:32]), .bs(2'd1), .mix(mix), .dec(1'b0), .rd(tempb));
	
aes32_2dir aesc(.rs1(key[0+:32]), .rs2(vector[0+:32]), .bs(2'd3), .mix(mix), .dec(1'b0), .rd(tempc));
aes32_2dir aesd(.rs1(tempc), .rs2(vector[32+:32]), .bs(2'd0), .mix(mix), .dec(1'b0), .rd(tempd));
aes32_2dir aese(.rs1(tempd), .rs2(vector[64+:32]), .bs(2'd1), .mix(mix), .dec(1'b0), .rd(tempe));
aes32_2dir aesf(.rs1(tempe), .rs2(vector[96+:32]), .bs(2'd2), .mix(mix), .dec(1'b0), .rd(tempf));

assign out = {temp3, temp7, tempb, tempf};
```

As such, I get the right cypherword but I fear this implementation might be non-standard which is not what I'm looking for.

PS: the MixColumns is replaced by
```
wire [31:0] mix_out_enc = {sb_out,     sb_out       ,    xtN(sb_out,4'd3 )      , xtN(sb_out,4'd2 )};
```

jnk0le, 2022-04-17:
> Let us imagine the following 128-bit word: 0x000102030405060708090a0b0c0d0e0f. Using little-endian conventions, the first 32-bit subword should be 0x0c0d0e0f, the second 0x08090a0b and so forth with [...]

FIPS 197 par. 3 standardizes this:

`0x00010203` is column 0
`0x00` - row 0 of col 0
`0x01` - row 1 of col 0

endianess matters when loading whole columns - rows reversed in registers

lvanderwiele, 2022-04-17:
That's the AES convention but the aes32esmi instruction does not seem to work that way.

According to the spec, we are supposed to have
```
let shamt : bits( 5) = bs @ 0b000; /* shamt = bs*8 */
let si : bits( 8) = (X(rs2)[31..0] >> shamt)[7..0];
```
So, assuming the 128-bit plaintext is split {t0, t1, t2, t3} and t0 is split {b0, b1, b2, b3}  when doing `aes32esmi a0, a0, t0, 0` we are working on the last 8 bits of t0 (b3) rather than b0.

At the moment, I'm trying to get the `tb_checker_saes32` from `rtl/crypto-fu/tb_checker_saes32.v` to work but I can't seem to get that either. The result of my aes block and the checker match but I can't arrange them to work properly to encrypt a 128-bit word (the reasoning that if I can chain 16 of them I can simulate the 16 instructions occurring) and I can't.

jnk0le, 2022-04-18:
>So, assuming the 128-bit plaintext is split {t0, t1, t2, t3} and t0 is split {b0, b1, b2, b3} when doing aes32esmi a0, a0, t0, 0 we are working on the last 8 bits of t0 (b3) rather than b0.

RISCV loads by default as little endian so `aes32esmi a0, a0, t0, 0` accesses row 0 (`t0 & 0x000000ff`)

lvanderwiele, 2022-04-22:
So shouldn't the instructions the first 32bit of the cypherword out of plaintext {t0, t1, t2, t3} and key {a0, a1, a2, a3} the following?
```
aes32esmi k0, k0, t0, 3
aes32esmi k0, k0, t1, 2
aes32esmi k0, k0, t2, 1
aes32esmi c0, k0, t3, 0
```
Because referring to your previous comment, if you have column 0 be `0x00010203` and `0x00` and `0x01` be rows 0 and 1 of that column, if `aes32esmi a0, a0, t0, 0` is indeed accessing `t0 & 0x000000ff`, that's accessing row 3 (`0x03`)

Or do I need to perform a `rev8` beforehand?

jnk0le, 2022-04-23:
>aes32esmi k0, k0, t0, 3
aes32esmi k0, k0, t1, 2
aes32esmi k0, k0, t2, 1
aes32esmi c0, k0, t3, 0

it's true for big endian riscv (no need for `rev8`), typically it's little endian so:

`0x000102030405060708090a0b0c0d0e0f` AES standard treats as array of bytes

loading first chunk gives `0x03020100` in register.

so in this case of doing:
```
la t0, cipherkey
lw a0, 0(t0)
addi a0, a0, 0x55
sw a0, 0(t0)
```
you get
`0x550102030405060708090a0b0c0d0e0f`

mjosaarinen, 2022-04-23:
> Hi everyone. I'm currently writing my master's thesis on RVK and encountering a bit of an issue with AES.

Hello! Cool stuff.

(..) 
> It would seem however that the code does not work as expected (at least not when comparing it to the example vectors found here: http://www.herongyang.com/Cryptography/AES-Example-Vector-of-AES-Encryption.html
> 
(..)
> Let us imagine the following 128-bit word: 0x000102030405060708090a0b0c0d0e0f. Using little-endian conventions, the first 32-bit subword should be 0x0c0d0e0f, the second 0x08090a0b and so forth with word thus being {w3, w2, w1, w0}. 

In the hex blobs referenced the bytes would be `00 01 02 03 .. 0f` and hence the first word, when read into a little-endian 32-bit system would have value `0x03020100`.

The instructions are designed to work without byte reversion on little-endian targets, and hence the word representations are different from FIPS 197 and other NIST standards where big endian / "network byte order" notation is used. Ours work with on-wire byte encodings of IETF protocols such as TLS and IPSec without byte swapping (again, assuming that the system is little endian -- for a big-endian RISC-V I'd recommend simply reversing the logic of AES instructions.)

So, these (or other) test vectors are not in little endian notion. Using the AES-128 in that example (which is also contained in FIPS 197 itself), you can use openssl on command line to create test vectors and see what the "real-life" byte ordering is.
```
# create a 16-byte file with a sequence of bytes 00 11 22 .. ff

echo -n 00112233445566778899aabbccddeeff | xxd -r -p - > tv.dat
hexdump -C tv.dat
00000000  00 11 22 33 44 55 66 77  88 99 aa bb cc dd ee ff  |.."3DUfw........|

# we can use openssl in command line to do the raw AES-CBC encryption:

cat tv.dat | openssl enc -aes-128-ecb -K 000102030405060708090a0b0c0d0e0f -nopad | hexdump -C
00000000  69 c4 e0 d8 6a 7b 04 30  d8 cd b7 80 70 b4 c5 5a  |i...j{.0....p..Z|
```

> I tried to get the first 32 bits of the cyphertext by applying the 4 first four aes instructions mentioned above. However, after doing so (with MixColumns disabled and the key set to 0x0 to be able [...]

The entire internal logic is reversed since RISC-V is usually a little-endian system. The word byte order in sources like FIPS 197 standard appendices ( https://doi.org/10.6028/NIST.FIPS.197 ) are reversed.

This can be done for AES but not for all algorithms:

Since AES does not need arithmetic operations with carry (add/sub), just XOR, we can keep things in this reverse order throughout, and hence eliminate the byte swap operations otherwise required for every block (this could be a 10% performance saving). There are other algorithms addressed by RVK that use additions and subtractions -- this is why SHA2 and SM3 implementations actually do require byte swap. SHA-3 is again carry-free so no byte swapping required, as is SM4 (where the big-endian rotation logic is encoded into the instructions specifically to avoid byte swapping).

You can find running code and test vectors at https://github.com/riscv/riscv-crypto/tree/master/benchmarks/aes and also (using intrinsics) at https://github.com/rvkrypto/rvkrypto-fips/tree/main/aes   There is testing code at https://github.com/rvkrypto/rvkrypto-fips/blob/main/test/test_aes.c

Cheers,
- markku

kdockser, 2022-10-28:
Closing as this is a general AES question and is not a change/issue related to the Crypto specifications.

closed 2022-10-28 by kdockser