riscv/riscv-crypto #159: LICENSE about RTL

ZenithalHourlyRate, 2022-05-16:
I recently tried to upstream the sbox implementation of `rtl/crypto-fu/riscv_crypto_fu_sboxes.v` to chisel . See https://github.com/chipsalliance/chisel3/pull/2533 and [the thread about the LICENSE issue](https://github.com/chipsalliance/chisel3/pull/2533#discussion_r873839963)

However, we are concerned about the LICENSE here. Overall the repo is `CC-BY-4.0 license` but this file stated that
`Copyright (c) 2020, PQShield Ltd. All rights reserved.` I do not know if I have the right to upstream it and what LICENSE I should use.

ZenithalHourlyRate, 2022-05-18:
Cc @mjosaarinen

mjosaarinen, 2022-05-18:
The copyright owner grants rights to use the work as stated in the license text. This does not mean that the copyright owner relinquishes the copyright itself. The license is "unlimited" and "for all time" but one needs to abide by its rules; for example, CC 4.0 requires you to keep the original copyright notices intact.

So whether or not this is a compatible license with your project is basically a question of whether you want to comply with CC 4.0.  

Happy that you've found it useful.

Cheers,
- markku

ZenithalHourlyRate, 2022-05-18:
> The copyright owner grants rights to use the work as stated in the license text.

Thanks for this clarification!

closed 2022-05-18 by ZenithalHourlyRate