riscv/riscv-crypto #171: Questions on vector crypto

nibrunie, 2022-08-13:
@kdockser and all, a few questions for the next meeting.

- Why use both `vs1` and `vs2` and concatenate them to hold 256-bit keys for AES-256 rather than using a widen element group ? (in a similar layout to the one used by narrowing shifts were one of the operand as 2*SEW and EMUL=2*LMUL ? what are the benefits of this approach ?
- I think the "round number" in https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vaesfk1.adoc could be replace by "round index". If I understood correctly, It is used to select the AES round constant, not to specify how many round of KeySchedule are executed (only one is executed)
- for the `vandn` instruction https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vandn.adoc, should an element group variant be added ? (in particular for the `.vx` and `.vi` variants, an element group variant could allow a different splat than what is allowed through SEW (since the max SEW `ELEN` may be smaller that the element group size which must be supported) ?

kdockser, 2022-08-18:
Thanks for the Questions, Nicolas. Here are my responses:

1. While AES-256 does have a 256-bit cipher key, the rounds keys (which include the cipher key) are all 128-bits wide. As you pointed out in today's TG, the round keys needed to generate the next round key might not be in the same register, yet the last two round keys are needed to generate the next round key. Therefore, we agreed to continue to use 128-bit wide element groups.
2. The original form of the forward round key generation instruction used a _round index_. This allowed it to work for both AES-128 and AES-256. When we decided that we also want to provide reverse round key generation instruction, we realized that the reverse instruction needs to know if it is generating for AES-128 or AES-256. For consistency, the forward instructions were changed to use this information and the _round number_ to determine the correct behavior rather than the index to the constant.
3. Interesting question on `vandn` and the element groups. This instruction was primarily added to provide a substantial performance boost to SHA3 code; I don't believe that it needs to have any special element-group properties to provide that speed up. Do others have insight on this?
4. I have added "internally" to the all-rounds instructions to clarify that the round keys are generated internally to the instruction and are not architecturally visible to other instructions.
5. I have updated `sha256c` to talk about `EGW=128`. There are still some consistency differences between the instruction descriptions; these will be resolved as we converge on a common scheme for all of the instructions' descriptions.

nibrunie, 2022-08-19:
Thank you @kdockser , this cover all my interrogations. I will let people add their opinions for 3. (a.k.a element group variant of `vandn` and possibly other instructions)

kdockser, 2022-10-28:
Closing as no one else has responded with their opinion on a need for other versions of 'vandn' that operates on element groups. I don't see a need for this right now; we can revisit it at a later date.

closed 2022-10-28 by kdockser